peted-davis/WeatherFlow_PiConsole #2: Version 3 errors out after a few minutes:

mattncsu, 2020-03-30:
Running on a Pi3 on Rasbian Stretch using official 7" touchscreen.  Let me know if there are any other logs that are helpful.

Thanks!
````
pi@raspberrypi-J:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_20-03-30_2.txt
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.5/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_pil, img_gif (img_sdl2, img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
2020-03-30 07:33:09-0500 [-] Log opened.
2020-03-30 07:33:24-0500 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0x7080ff30>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event3
[INFO   ] [MTD         ] Read event from </dev/input/event3>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [MTD         ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event4
[INFO   ] [MTD         ] Read event from </dev/input/event4>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [MTD         ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event3
[INFO   ] [HIDInput    ] Read event from </dev/input/event3>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [HIDInput    ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event4
[INFO   ] [HIDInput    ] Read event from </dev/input/event4>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [HIDInput    ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event3> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event4> range position X is 0 - 800
[INFO   ] [MTD         ] </dev/input/event1> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event2> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event4> range position Y is 0 - 480
[INFO   ] [HIDMotionEvent] using <HL 0000 00 00 00 01 Smart Control>
[INFO   ] [MTD         ] </dev/input/event1> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] using <HL 0000 00 00 00 01 Smart Control System Control>
[INFO   ] [MTD         ] </dev/input/event2> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 0
[INFO   ] [HIDMotionEvent] using <FT5406 memory based driver>
[INFO   ] [MTD         ] </dev/input/event4> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] using <HL 0000 00 00 00 01 Smart Control Consumer Control>
[INFO   ] [Support     ] Starting twisted reactor
[INFO   ] [MTD         ] </dev/input/event1> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event3> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] using <HL 0000 00 00 00 01 Smart Control>
[INFO   ] [MTD         ] </dev/input/event2> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range ABS X position is 0 - 800
[INFO   ] [MTD         ] </dev/input/event4> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event4> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event3> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event2> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range ABS Y position is 0 - 480
[INFO   ] [MTD         ] </dev/input/event1> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event4> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event3> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event2> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range position X is 0 - 800
[INFO   ] [MTD         ] </dev/input/event1> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event4> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event2> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range position Y is 0 - 480
[INFO   ] [MTD         ] </dev/input/event1> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event2> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Base        ] Leaving application in progress...
[INFO   ] [Support     ] Stopping twisted threads
[INFO   ] [Support     ] Shutting down twisted reactor
2020-03-30 07:43:08-0500 [-] Traceback (most recent call last):
2020-03-30 07:43:08-0500 [-]   File "main.py", line 908, in <module>
2020-03-30 07:43:08-0500 [-]     wfpiconsole().run()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/app.py", line 855, in run
2020-03-30 07:43:08-0500 [-]     runTouchApp()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/base.py", line 504, in runTouchApp
2020-03-30 07:43:08-0500 [-]     EventLoop.window.mainloop()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/core/window/window_egl_rpi.py", line 94, in mainloop
2020-03-30 07:43:08-0500 [-]     self._mainloop()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/core/window/window_egl_rpi.py", line 89, in _mainloop
2020-03-30 07:43:08-0500 [-]     EventLoop.idle()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/base.py", line 339, in idle
2020-03-30 07:43:08-0500 [-]     Clock.tick()
2020-03-30 07:43:08-0500 [-]   File "/home/pi/.local/lib/python3.5/site-packages/kivy/clock.py", line 591, in tick
2020-03-30 07:43:08-0500 [-]     self._process_events()
2020-03-30 07:43:08-0500 [-]   File "kivy/_clock.pyx", line 384, in kivy._clock.CyClockBase._process_events
2020-03-30 07:43:08-0500 [-]   File "kivy/_clock.pyx", line 414, in kivy._clock.CyClockBase._process_events
2020-03-30 07:43:08-0500 [-]   File "kivy/_clock.pyx", line 412, in kivy._clock.CyClockBase._process_events
2020-03-30 07:43:08-0500 [-]   File "kivy/_clock.pyx", line 167, in kivy._clock.ClockEvent.tick
2020-03-30 07:43:08-0500 [-]   File "/home/pi/wfpiconsole/lib/forecast.py", line 185, in <lambda>
2020-03-30 07:43:08-0500 [-]     Clock.schedule_once(lambda dt: self.DownloadForecast(),600)
2020-03-30 07:43:08-0500 [-] NameError: name 'self' is not defined
````

peted-davis, 2020-03-30:
Can you let me know you Station ID? I think I have fixed the bug but want to run some tests.

mattncsu, 2020-03-30:
9457

mattncsu, 2020-03-30:
I also haven't been getting a forecast if that plays in to it at all. I haven't had a chance to dig in to why yet.

peted-davis, 2020-03-30:
Yes, it is associated with missing forecast. The DarkSky forecast is failing, and there is a bug when it tries to get the forecast again 10 minutes later

mattncsu, 2020-03-30:
FYI, aviationweather.gov has an API access to METAR and TAF data. I don't think any access keys are required. Not sure if it is US only or world wide

peted-davis, 2020-03-30:
OK I've patched the bug. You can update your install by running the command `wfpiconsole patch`. Let me know how you get on.

mattncsu, 2020-03-30:
Patched, I'll let you know if there are any further errors. Is the lack of forecast a DarkSky issue?

peted-davis, 2020-03-30:
Yes, it seems like a DarkSky issue. It was working for me sometimes and not at other times while I was testing the patch

mattncsu, 2020-03-30:
So far, so good after an hour. Thanks for the quick fix!

closed 2020-03-30 by mattncsu