peted-davis/WeatherFlow_PiConsole #4: Reboot and Exit Console option in settings

AeroSavvy, 2020-05-05:
Hello! I really enjoy using my console. There are times that I would like to exit the console or do a reboot without having to get on my laptop and use PuTTY.

I have auto start enabled, so without using PuTTY, I don't see an easy way to exit the console. 

Would it be difficult to add Reboot and Exit Console options in the settings?

Thanks!

peted-davis, 2020-05-06:
Hi! Glad you are enjoying using the console. Adding a reboot option direct from the console would be hard, as it requires elevated permissions to run a reboot command. An exit option should be possible though, and I will add this to the to-do list. At least with an exit option you could then reboot using the Rasbian desktop GUI.

AeroSavvy, 2020-05-06:
The exit option would be perfect! I forgot that I can easily reboot from the GUI. Thank you so much.

peted-davis, 2020-06-08:
Reboot and shutdown buttons have been added to Version 3.5

closed 2020-06-08 by peted-davis