peted-davis/WeatherFlow_PiConsole #5: openweathermap.org

ctgreybeard, 2020-06-08:
I have added support for openweathermap.org in my fork. It needs validating by someone else. I just got my Tempest yesterday so I'm still learning about it.

I did notice one other, unrelated, thing ... the buttons at the bottom of the window are very tiny in the png file and so when they get sized they become extremely blurry. I haven't tried it on my Pi4 yet but it  doesn't look good  on my macbook.

peted-davis, 2020-06-08:
Nice work incorporating openweathermap. I will officially be adding support for the WeatherFlow generated forecast in the coming weeks when the API has been finalised.

Sorry the buttons look bad on your macbook. The console has been specifically designed for the Raspberry Pi, so mileage will vary on higher resolution screens.

ctgreybeard, 2020-06-08:
Thanks. Due to the way you structured the code it wasn't especially difficult.  But speaking of the Pi ... I installed it on my Pi 4 with the  7" screen and find that the GUI is shifted up too far and to the left.

Here's a screenshot with scrot but that  isn't what I am actually seeing ...

[image]

Here a shot with a camera:

[image]

Unfortunately without the title bar I don't know how to drag it back into position.

I'm more than happy to poke around if you could give me an idea of where  I might look ...

And, as you  can see, the buttons are fine  on the Pi.

ctgreybeard, 2020-06-08:
Here's a shot after I updated to  your latest and  applied my Openweathermap code ...

[image]

peted-davis, 2020-06-08:
Have you followed the instructions in the README for hiding the taskbar on the Raspberry Pi 4? I think that will fix your issue:

**Raspberry Pi 4**

For those of you running a Raspberry Pi 4 an extra step is needed to get the console running perfectly. The “Task Bar” panel on the Raspberry Pi desktop needs to be hidden or else it will displace the console on the screen. There are two options to achieve this. First right click on the “Task Bar” panel on the Raspberry Pi desktop and select Panel settings. Select the Advanced tab. Then either:

Un-tick `Reserve space, and not covered by maximised windows`, or

Tick `Minimise panel when not in use` and set `Size when minimised` to 0 pixels.

ctgreybeard, 2020-06-08:
> 
> Un-tick `Reserve space, and not covered by maximised windows`, or
> 
> Tick `Minimise panel when not in use` and set `Size when minimised` to 0 pixels.
> 
> 

Neither one of those suggestions changed the problem.

ctgreybeard, 2020-06-09:
Found it ... my error ... when I copied the .ini file it still had "other" for the system type. Changing it to Pi4 and it looks wonderful!

Are you interested in a pull request for the Openweathermap changes or will the new Weatherflow forecast  be good? One thing I noted about the Openweathermap forecast was that it  did not include Precip Prediction Percent which I found puzzling and I think it's an important metric.

peted-davis, 2020-06-09:
> Found it ... my error ... when I copied the .ini file it still had "other" for the system type. Changing it to Pi4 and it looks wonderful!

That would do it! Glad you managed to track down the issue.

> Are you interested in a pull request for the Openweathermap changes or will the new Weatherflow forecast be good?

I think I will stick with the WeatherFlow forecast for the moment as I am trying to reduce the number of different API providers the console relies on.

ctgreybeard, 2020-06-09:
> I think I will stick with the WeatherFlow forecast for the moment as I am trying to reduce the number of different API providers the console relies on.

Understood. I haven’t looked but I was wondering if you capture the UDP broadcast messages? There is some interesting stuff in there.

peted-davis, 2020-06-10:
No, the console relies entirely on the WeatherFlow websocket and REST API. This means it misses out on some of the interesting UDP data that is not available elsewhere, but the quality of the observations in the websocket/API data is better due to the post-processing carried out on the WeatherFlow servers (see: https://community.weatherflow.com/t/lightning-reports-tempest/6262/25)

closed 2020-06-11 by ctgreybeard