peted-davis/WeatherFlow_PiConsole #7: Exception calling time.sleep in  station.py

ctgreybeard, 2020-06-12:
In station.py there are calls to time.sleep but `time` is `datetime.time` which does not have `sleep`. You don't appear to be using `datetime.time` at all.

```
2020-06-11 20:18:34-0400 [-] Log opened.
2020-06-11 20:18:34-0400 [-] https://api.openweathermap.org/data/2.5/onecall?appid=aa8c27b8a59a4ba3487556a450b6e720&lat=41.38648&lon=-73.38995&exclude=current,minutely,daily&units=metric
2020-06-11 20:18:35-0400 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0xaceaeb30>
2020-06-11 20:18:36-0400 [-] Websocket connection open
2020-06-11 20:18:54-0400 [-] Exception in thread getObservationCount:
2020-06-11 20:18:54-0400 [-] Traceback (most recent call last):
2020-06-11 20:18:54-0400 [-]   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
2020-06-11 20:18:54-0400 [-]     self.run()
2020-06-11 20:18:54-0400 [-]   File "/usr/lib/python3.7/threading.py", line 865, in run
2020-06-11 20:18:54-0400 [-]     self._target(*self._args, **self._kwargs)
2020-06-11 20:18:54-0400 [-]   File "/home/pi/Projects/WeatherFlow_PiConsole/lib/station.py", line 185, in getObservationCount
2020-06-11 20:18:54-0400 [-]     time.sleep(0.01)
2020-06-11 20:18:54-0400 [-] AttributeError: type object 'datetime.time' has no attribute 'sleep'
2020-06-11 20:18:54-0400 [-]
```

I removed the import for `datetime.time` and inserted an import for `time` and that eliminated the error. That did not fix my problem of not getting any data but I'm  working on that ...

peted-davis, 2020-06-12:
Good spot! That is definitely a bug. It should be `import time` instead of `from datetime import time`. I will fix this in the next version.

peted-davis, 2020-06-19:
This should now be fixed in Version 3.51

closed 2020-06-19 by peted-davis