peted-davis/WeatherFlow_PiConsole #8: Pi4 touchscreen mouse “clicks” not recognized

ctgreybeard, 2020-06-17:
I have a new Pi4 that I set up on and have a strange problem. The app doesn’t recognize the “clicks” when using the touchscreen. It does fine when I connect a USB mouse. And I can control the desktop using only the touchscreen so the problem only seems to exist in the app.

I have a different Pi4 with the exact same hardware setup and it works fine so I suspect it’s a package missing or a config option that I have set on the newer Pi. 

There aren’t any errors in the log although I haven’t tried using debug yet. I thought I would ask here first.

Is there a list of dependancies that I can check against or do you have any settings I can check?

peted-davis, 2020-06-17:
That's weird. The Kivy config file needs to be edited to enable the touchscreen, but the installer should have taken care of that. 

Follow the instructions under ”Using Official RPi touch display” on this page https://kivy.org/doc/stable/installation/installation-rpi.html and see if that helps.

ctgreybeard, 2020-06-17:
Adding the two lines to the kivy config.ini did it. The mouse line was already there and one other line. I ended up with

```
[input]
mouse = mouse
%(name)s = probesysfs,provider=hidinput
mtdev_%(name)s = probesysfs,provider=mtdev
hid_%(name)s = probesysfs,provider=hidinput
```
And it is working correctly. I will go over the rest of the manual install later to be sure there isn’t anything else I might have missed.

peted-davis, 2020-06-18:
Great! Glad that was the fix. I will have a look at the install code and see if there are any issues with it.

closed 2020-06-18 by peted-davis

ctgreybeard, 2020-06-23:
I wouldn’t bother, I was rolling my own installation.