peted-davis/WeatherFlow_PiConsole #10: Add local temperature/humidity sensor?

ctgreybeard, 2020-06-22:
I have the software running on a Pi 4 with the 7" LCD touchscreen and all is working very well. One thing I am missing is indoor temperature. I have an Adafruit AHT20 sensor that is a small I2C device. I have tested it on another machine and it works great. The interface is python using the Adafruit Blinka library and one device module. Reading the values could not be simpler.

My question is on where to integrate that into the console. I am a software guy, fluent in python, so I am sure I can do it but if you could give me an idea of the "best" starting point I would appreciate it.

peted-davis, 2020-06-23:
To integrate the sensor into the console, you first need to write a function (either in main.py or as a separate module) that retrieves the temperature from AHT20. The exact details of how that is done doesn’t matter, but make sure it is a non-blocking process. Once you have that function, you can use `clock.schedule_interval` to call the function at set intervals (I would suggest every 60 seconds to match the Tempest data). If you look on lines 273-275 of `main.py` you will see examples of this. For example, `astro.sunTransit` is called every second to calculate the position of the sun icon on the day/night bar.

Once you have a function being called regularly, you need to assign the output to variables that are displayed on the console. You can’t use any old variable. Instead you have to use a Kivy property. Doing this ensures that a change in the value of the variable results in the console display being updated. You will see on line 200 of `main.py` that I define `Obs` as a `DictProperty` and in that dictionary is a field defined as `('inTemp','--')`. You can use this variable to display your AHT20 temperature by setting index 0 to the temperature (as a string) and index 1 to the unit (as a string). You will need to pass the whole `Obs` `DictProperty` to the function you made earlier as an input such that you can set the values of `inTemp` as required, and make sure you return it at the end. Again checkout the `astro.sunTransit` function as it is very similar to what you want to do.

Just to add - the layout of the console is defined in `wfpiconsole.kv`. If you look on lines 262-272 you can see how the values of `inTemp` are added to the layout.

ctgreybeard, 2020-06-23:
Thank you very much, that is exactly what I need to make it happen.

ctgreybeard, 2020-06-23:
A question: in wfpiconsole.kv it check for app.IndoorTemp set to "1" to show the indoor temp but I don't see how that is set.

peted-davis, 2020-06-23:
That is set in the settings under the 'Display' tab, and the variable is defined on line 228 in `main.py`. You can choose to hide the Indoor temperature if you wish. As long as you leave the indoor temperature as set to `on` in the settings, then `app.IndoorTemp` is set to 1 everything should work as expected. Remember you want to store your indoor temperature in the 'inTemp' field in the 'Obs' dictionary. You don't want to be storing it in the 'app.IndoorTemp' variable.

ctgreybeard, 2020-06-24:
I have to hook up the hardware tomorrow but I think I have it coded.  I've never worked with kivy so this is fun. I did notice an anomaly that appears to work just because "python" ... 

In the Obs init you use `('inTemp','--')` which assigns  a default to `inTemp` as the two-character string `'--'` but ... when kivy references `inTemp` it uses it as a list getting `[0]` and `[1]` ... it seems to me that this works only because strings are usable that way, `[0]` is `'-'`, etc. Am I right in this or have I missed something? And, related, I suppose that if I don't get a temp from my board for whatever reason I can simply use `'--'` as the value to set `inTemp` to?

P.S. Thank you for teaching me `enumerate()` ... I had never used that before.

peted-davis, 2020-06-24:
Yep, that's just Python and it just works for strings. `[0]` refers to the first `-`, and `[1]` refers to the second `-`. The `-`s can also be replaced by other strings by using the `[0]` and `[1]` indexes. You may see that in other places I have had to initiate variables as a list of strings (e.g. `'Moonrise',['-','-']`). This is because I replace one of the `-` with an object (like a `datetime` object), and therefore the shorthand `--` doesn't work. 

> And, related, I suppose that if I don't get a temp from my board for whatever reason I can simply use '--' as the value to set inTemp to?

Yes, that is what I do. (well I initially set it to NaN, which then gets converted to `-` during the unit conversion routines. You can probably ignore this complication as you won't really need to change the indoor units).

ctgreybeard, 2020-06-24:
OK! It's all working very well. Here's a couple of shots. The first is the screen with the indoor temp with max and min and the other is the sensor that I'm using. It's the [Adafruit AHT20](https://www.adafruit.com/product/4566) I need to make a bracket for the sensor. It does need to stay away from the Pi, though. I found that the temp was about four degrees high when it was sitting on top of the console.

[image]
[image]

closed 2020-06-24 by ctgreybeard