peted-davis/WeatherFlow_PiConsole #9: Websocket Errors

Zwilrich, 2020-06-22:
I just installed wfpiconsole and I am getting numerous websocket errors.
Background information:
- Installed per the instructions, using `curl`.
- Running version 3.51.
- `cat /proc/device-tree/model:  Raspberry Pi 3 Model B Plus Rev 1.3`
- The ini file shows Hardware = Pi3
- `uname -a:  Linux pi-touch 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux`
- My station is a Tempest.
- There are entries for TempestID and TempestHeight in the .ini file.
- Running with an HDMI monitor until I connect the touchscreen.
- I have a working DarkSky key.
- I have not requested a WeatherFlow API key.  However, there is one in the .ini file.

My ssh console shows:
>wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_20-06-22_1.txt
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil, img_gif (img_ffpyplayer ignored)
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
2020-06-22 15:28:55-0600 [-] Log opened.
2020-06-22 15:28:56-0600 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [Support     ] Starting twisted reactor
[INFO   ] [GL          ] NPOT texture support is available
2020-06-22 15:28:57-0600 [-] failing WebSocket opening handshake ('WebSocket connection upgrade failed (400 - BadRequest)')
2020-06-22 15:28:57-0600 [-] dropping connection to peer tcp4:35.171.114.243:80 with abort=True: WebSocket connection upgrade failed (400 - BadRequest)
2020-06-22 15:28:57-0600 [-] Websocket connection closed
2020-06-22 15:28:57-0600 [-] Websocket connection retrying
2020-06-22 15:28:57-0600 [-] <twisted.internet.tcp.Connector instance at 0x6fe6f430 disconnected IPv4Address(type='TCP', host='ws.weatherflow.com', port=80)> will retry in 2 seconds
2020-06-22 15:28:57-0600 [-] Stopping factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:00-0600 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:00-0600 [-] failing WebSocket opening handshake ('WebSocket connection upgrade failed (400 - BadRequest)')
2020-06-22 15:29:00-0600 [-] dropping connection to peer tcp4:107.22.116.161:80 with abort=True: WebSocket connection upgrade failed (400 - BadRequest)
2020-06-22 15:29:00-0600 [-] Websocket connection closed
2020-06-22 15:29:00-0600 [-] Websocket connection retrying
2020-06-22 15:29:00-0600 [-] <twisted.internet.tcp.Connector instance at 0x6fe6f430 disconnected IPv4Address(type='TCP', host='ws.weatherflow.com', port=80)> will retry in 7 seconds
2020-06-22 15:29:00-0600 [-] Stopping factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:08-0600 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:08-0600 [-] failing WebSocket opening handshake ('WebSocket connection upgrade failed (400 - BadRequest)')
2020-06-22 15:29:08-0600 [-] dropping connection to peer tcp4:35.171.114.243:80 with abort=True: WebSocket connection upgrade failed (400 - BadRequest)
2020-06-22 15:29:08-0600 [-] Websocket connection closed
2020-06-22 15:29:08-0600 [-] Websocket connection retrying
2020-06-22 15:29:08-0600 [-] <twisted.internet.tcp.Connector instance at 0x6fe6f430 disconnected IPv4Address(type='TCP', host='ws.weatherflow.com', port=80)> will retry in 20 seconds
2020-06-22 15:29:08-0600 [-] Stopping factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:28-0600 [-] Starting factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
2020-06-22 15:29:28-0600 [-] failing WebSocket opening handshake ('WebSocket connection upgrade failed (400 - BadRequest)')
2020-06-22 15:29:28-0600 [-] dropping connection to peer tcp4:107.22.116.161:80 with abort=True: WebSocket connection upgrade failed (400 - BadRequest)
2020-06-22 15:29:28-0600 [-] Websocket connection closed
2020-06-22 15:29:28-0600 [-] Websocket connection retrying
2020-06-22 15:29:28-0600 [-] <twisted.internet.tcp.Connector instance at 0x6fe6f430 disconnected IPv4Address(type='TCP', host='ws.weatherflow.com', port=80)> will retry in 54 seconds
2020-06-22 15:29:28-0600 [-] Stopping factory <__main__.WeatherFlowClientFactory object at 0x6fe6f390>
^C

peted-davis, 2020-06-23:
I haven't seen this before. Does the console ever connect, or do you always get these errors? How is your home network setup? Is it more complex than a single router and WiFi network? 

Line 378 in `main.py` should currently contain: `reactor.connectTCP('ws.weatherflow.com',80,self._factory,)`. Can you replace it with `reactor.connectSSL('ws.weatherflow.com',443,self._factory,ssl.ClientContextFactory(),20)` and let me know if that helps?

Zwilrich, 2020-06-23:
The console appears but the only thing it shows is the forecast.  The errors continue until I get tired of seeing them.  There is nothing unusual with my network setup.  However, I have had issues with my stupid ISP (CenturyLink) rewriting the request headers from HTTP 1.1 to HTTP 1.0.
I'll try changing the port from 80 to 443 and the connect from TCP to SSL later today and let you know what happens.  If that doesn't fix it, I'll try using a VPN.

Zwilrich, 2020-06-23:
Sorry, I didn't to close this issue.

Zwilrich, 2020-06-23:
Changing the websocket to a secure connection fixed the issue.  I'll blame CenturyLink for the problem.
I'll let you close this issue in case you have any additional comments.
Thanks for a great program.

peted-davis, 2020-06-24:
Great! Glad that fixed it. tbh the connection was always meant to be secure. I must have rolled back to the un-secured version for some testing and then forgot to revert back to the secure version before pushing to the master branch. I will make sure that I revert to the secure connection in the next version.

closed 2020-06-24 by peted-davis