peted-davis/WeatherFlow_PiConsole #11: Wfpiconsole disconnects from tempest

Xadamx1, 2020-06-23:
Hi,
Some background info, running on a pi3. Installed using the bash script. I am using the touch screen.

The console works great when it first starts up. However after about 2 hours it will disconnect from my Tempest and refuse to reconnect. I have to either restart the pi or issue the wfpiconsole stop command and restart it to connect again.

I still have connectivity through the tempest app so I know the weather station did not go down, seems to be something with the console itself.

peted-davis, 2020-06-23:
Hi,
I need more info to investigate this. Are you seeing any errors being printed to the terminal, or can you copy and paste the contents of `wfpiconsole.log` here? When it disconnects does the clock on the console stop as well, or does that continue? Can you double check all of your API keys in `wfpiconsole.ini` and make sure there are no errors.

Xadamx1, 2020-06-23:
Hi,
Thanks for the quick reply, upon further investigation as per your instructions it would appear the problem is with my Pi staying connected to the wifi. Probably has nothing to do with you at all. I will investigate. 

Thanks for making this console its perfect for what I was looking for. 
Also, in the Weatherflow app they have a forecast in there, is it possible to draw from that instead of dark sky?

closed 2020-06-23 by Xadamx1

peted-davis, 2020-06-23:
It isn't possible to use the WeatherFlow forecast at the moment, but it will be. They are still finalising the details of the public API, but as soon as that is done I will switch the console over