peted-davis/WeatherFlow_PiConsole #12: Using openweathermap.org as a replacement for darksky?

rmeitz, 2020-06-23:
Have you looked at openweathermap.org as a replacement for darksky?  They even have a link on their API page on migrating from darksky. I can look into this if you want to give me some pointers on where to look for the relevant code.

peted-davis, 2020-06-23:
Thanks for the offer to help, but I already have a plan in place to replace Dark Sky with a forecast provided directly from WeatherFlow (the same one as in the Tempest app). I am just waiting for WF to finalise the details on the public facing API, and I then I should be able to incorporate it into the console.

rmeitz, 2020-06-24:
Ok thanks. I will wait for that I guess. Do you have an estimate from them on when it will be available?
Closing the issue.

closed 2020-06-24 by rmeitz