peted-davis/WeatherFlow_PiConsole #13: Wind Speed | Direction Issue

Zwilrich, 2020-06-24:
The circle with the triangle showing the wind direction is frequently overlaid by a black square with direction in degrees, the wind speed, and a verbal direction (Due South).  The square is completely black, except for the text, and completely overlaps the circle except for one scan line at the top and one at the bottom.
Also, the predominant direction (Direction: N) appears on the east part of the circle instead of the lower-right corner of the block.  This happens rarely.

peted-davis, 2020-06-24:
The black box is an odd bug that crops up every now and again for some people and can usually be fixed by increasing the shared graphics memory. To do that, open a terminal and enter `sudo raspi-config`. Navigate to `Advanced options` -> `Memory split` and increase the value here to 256. Save the changes and reboot your pi and all should be fixed. 

> Also, the predominant direction (Direction: N) appears on the east part of the circle instead of the lower-right corner of the block. This happens rarely.

I can't picture what is going on here. Do you have a screen grab example?

Zwilrich, 2020-06-26:
Increasing the graphics memory seems to have fixed the problem.

As for the predominant direction appearing on the circle, that seems to have gone away after restarting the program.

I had a similar issue today where other parts of the screen were in the wrong places.  I think it was caused by a temporary loss of the websocket connection (or maybe the connection loss was a symptom of some other problem).  Either way, I don't think it is something that can be tracked and fixed.  Therefore, I am closing this issue.

Thanks again for a great program and the great support.

closed 2020-06-26 by Zwilrich