peted-davis/WeatherFlow_PiConsole #14: Forecast Limited to next hour

rcrum003, 2020-07-06:
This may be a personal preference, but can we make the forecast for the day, instead of the hour?

For example, when I leave the house in the morning, seeing what the forecast high/low helps me prepare for the day.
Seeing the high/low for the next hour isn't useful.

dethompson1958, 2020-07-06:
Yes, looks good but a 12 or 24 hour would be more useful.

Sent from my Verizon, Samsung Galaxy smartphone

rcrum003, 2020-07-06:
Yes, 12 or 24 would be good.

peted-davis, 2020-07-07:
Thanks for the suggestion. I am planning on revisiting the layout of the forecast panel soon, and will incorporate details from the daily forecast alongside the hourly values.

dethompson1958, 2020-07-07:
Thanks.

Sent from my Verizon, Samsung Galaxy smartphone

peted-davis, 2020-10-21:
A new forecast panel with both hourly and daily values in now available in Version 4.0

closed 2020-10-21 by peted-davis