peted-davis/WeatherFlow_PiConsole #16: Data retrieval fails after apt-update

Death2Beavis, 2020-07-28:
Just a note. I am current on 3.6 version of the console and all updates/upgrades to buster. I ran my normal update/upgrade process this morning and it broke the data retrieval from my Tempest. Sager, moon etc data was all good, data was blank. To fix it I had to rerun the wfpiconsole update command and reboot the pi. I am guessing this may have had something to do with the python manager package (only update that ran when updating the console....even though it was current I theorized that it might find something in its checks). Its working now, but this would be a concern.

Thanks.

peted-davis, 2020-07-29:
Interesting.. Was the console running while you ran your normal update/upgrade process, or did you stop it first? To fix it you say you had to rerun the `wfpiconsole update` command. Did you try stopping and restarting the console first? I am wondering whether the update/upgrade process updated a package that the console relied on, and that caused it to stop updating.

Death2Beavis, 2020-07-29:
Yes the console was running when I SSH'd into it, ran usual sudo apt
update/upgrade. I rebooted after that....no data. Stopped and restarted the
console. No data. That was when I had the thought to re run the wfpiconsole
update.  It went through its checks and updated the python package manager,
restarted. All good.

restackhouse, 2020-07-29:
I am doing a new install and I found moving to /tmp cleared up a permssion issue

peted-davis, 2020-07-29:
> Yes the console was running when I SSH'd into it, ran usual sudo apt update/upgrade. I rebooted after that....no data. Stopped and restarted the console. No data. That was when I had the thought to [...]

OK, understood. I'm not entirely sure what happened here, but it sounds like the update/upgrade messed with something. Although I'm not entirely sure why just updating the Python package manager would fix that... When the console is running it logs output to `wfpiconsole.log` in the `wfpiconsole` directory. If this happens again can you have a look at that file and let me know if you see any errors? It is overwritten whenever the console is restarted so remember to inspect it before you start the console again.

Death2Beavis, 2020-07-29:
RGR will do

closed 2020-09-17 by peted-davis