peted-davis/WeatherFlow_PiConsole #17: Tempest

restackhouse, 2020-07-29:
I have a new tempest (it rocks) when I try to install this console. it asks for my stationID, then says "Is this a tempest" and then asks if I have an AIR indoor -- of course the NEW tempest is all-in-one and there is not additional device.

Am I doing something wrong, or is the new tempest just too new?

Thanks in advance.

peted-davis, 2020-07-29:
Nope, you're not doing anything wrong. The console is completely compatible with Tempest, but it also works with the older generation Air/Sky hardware. Some users have an older Air module indoors, hence the prompt. You just need to answer 'No' to that question. You can then choose to hide the "Indoor Temperature" field in the console settings.

restackhouse, 2020-07-29:
confirmed! thanks.. I just noticed UNDER the ST-0000xxxxx number is a device ID

peted-davis, 2020-07-29:
Glad you're up and running.

>I just noticed UNDER the ST-0000xxxxx number is a device ID

Don't confuse the serial number of your Tempest with it's device ID as they are different! The device ID is assigned when you register the hub and Tempest for the first time using the WeatherFlow App. You can see the device ID in the app by clicking on `Settings` -> `Stations` -> '[station name]` -> `Status` and scrolling down to the Tempest section

closed 2020-08-05 by peted-davis