peted-davis/WeatherFlow_PiConsole #18: Occasional crashes from empty (perhaps) forecast?

ctgreybeard, 2020-08-04:
I am seeing a crash occasionally that appears to be because of empty forecast data. I haven't dug into it yet but I did want to post the problem.

The end of the log reads

```
2020-08-03 20:05:02-0400 [-] Traceback (most recent call last):
 2020-08-03 20:05:02-0400 [-]   File "main.py", line 1048, in <module>
 2020-08-03 20:05:02-0400 [-]     wfpiconsole().run()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/app.py", line 855, in run
 2020-08-03 20:05:02-0400 [-]     runTouchApp()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/base.py", line 504, in runTouchAp p
 2020-08-03 20:05:02-0400 [-]     EventLoop.window.mainloop()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line  747, in mainloop
 2020-08-03 20:05:02-0400 [-]     self._mainloop()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line  479, in _mainloop
 2020-08-03 20:05:02-0400 [-]     EventLoop.idle()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/base.py", line 339, in idle
 2020-08-03 20:05:02-0400 [-]     Clock.tick()
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/.venvs/wx/lib/python3.7/site-packages/kivy/clock.py", line 591, in tick
 2020-08-03 20:05:02-0400 [-]     self._process_events()
 2020-08-03 20:05:02-0400 [-]   File "kivy/_clock.pyx", line 384, in kivy._clock.CyClockBase._process_events
 2020-08-03 20:05:02-0400 [-]   File "kivy/_clock.pyx", line 414, in kivy._clock.CyClockBase._process_events
 2020-08-03 20:05:02-0400 [-]   File "kivy/_clock.pyx", line 412, in kivy._clock.CyClockBase._process_events
 2020-08-03 20:05:02-0400 [-]   File "kivy/_clock.pyx", line 167, in kivy._clock.ClockEvent.tick
 2020-08-03 20:05:02-0400 [-]   File "main.py", line 467, in UpdateMethods
 2020-08-03 20:05:02-0400 [-]     forecast.Download(self.MetData,self.config)
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/Projects/WeatherFlow_PiConsole/lib/forecast.py", line 52, in Download
 2020-08-03 20:05:02-0400 [-]     Extract(metData,Config)
 2020-08-03 20:05:02-0400 [-]   File "/home/pi/Projects/WeatherFlow_PiConsole/lib/forecast.py", line 85, in Extract
 2020-08-03 20:05:02-0400 [-]     metDict = metDict[bisect.bisect(Times,int(time.time()))]
 2020-08-03 20:05:02-0400 [-] IndexError: list index out of range
```

and the full log is attached.

[wxlog.zip](https://github.com/peted-davis/WeatherFlow_PiConsole/files/5019821/wxlog.zip)

peted-davis, 2020-08-05:
Thanks for the heads up. I will get a fix for this in place in the next version.

peted-davis, 2020-08-30:
Fixed in v3.7

closed 2020-08-30 by peted-davis