peted-davis/WeatherFlow_PiConsole #19: Sager Forecast ERROR: Missing METAR information

creas002, 2020-08-19:
I'm a new Tempest owner and recently setup wfpiconsole . Very nicely done Peter! Thank you for all your work to make this excellent console.

I'm getting the following panel error on my Pi 4b --> Sager Forecast ERROR: Missing METAR information. 

I seem to have all functionality working as expected except for this problem and I've been looking at it for a while. Hoping you can tell me how to debug this error. I'm attaching  ini and log files, which I don't notice any issue with.

[wfpiconsole.zip](https://github.com/peted-davis/WeatherFlow_PiConsole/files/5099846/wfpiconsole.zip)

peted-davis, 2020-08-20:
Thanks for your kind words! It looks like you are getting this error because the CheckWX API key in your .ini file is not correct. When I try and use it to retrieve the closest METAR to your station location I get the error:
 
`{"statusCode":401,"error":"Unauthorized","message":"invalid x-api-key in request header"}`. 

Can you login to your CheckWX account again (https://www.checkwxapi.com/auth/login) and double check that you have copied the API key correctly into the .ini file. If you have, you could try deleting the existing API key and generate a new one, or delete your account entirely (under the "Profile" tab) and start again. Hopefully one of these will get you up and running with a working API key. If nothing works let me know and we can troubleshoot further.

creas002, 2020-08-20:
Wow, that was it. Originally I wrote the API key down and manually typed it in. There was 2 repeating characters that I missed. I enabled ssh demon last night so making the change from SecureCRT was quick and painless.  I really struggled with determining the cause of the error. I read where you were reading about Sager and discussing with WeatherMan - Brian, but I never understood how you actually pulled it off. I did not see any signs of an error in the wfpiconsole.log. Wonder when it first initializes if if could log PASS/FAIL for testing the API key works. 

Thank you sir for your prompt reply and helping me sort this. I like giving a best effort before reaching out for help, but I was totally lost as to what was happening.

Kind Regards
Richard 
73 de AI4Y

peted-davis, 2020-08-20:
Great! Glad to hear you are up and running. That's a good idea to check whether the API key is valid. I should be able to do that easily during the initial generation of the config file so I will look into it.