peted-davis/WeatherFlow_PiConsole #23: Temperature and Wind Speed panes with no or incorrect data

dwburger, 2020-09-08:
I was able to install and start this cool project in no time due, primarily, to the excellent installation instructions and one-command process.  I've been watching the console for a few hours and noticed that the upper-middle "Temperature/Outdoor" pane and the upper-right "Wind Speed | Direction" pane display no data or incorrect data.  The upper-left "Current Forecast" pane displays the correct outdoor temperature and wind speed/direction, but those values are not displayed in the other panes (see photo below).  Is there something more I need to do to get the other panes to work properly???  Thanks so much for making this Python console for the WeatherFlow Tempest available!

[image]

falkyre, 2020-09-09:
Are your logs showing any exceptions being thrown?  Is this a new install of a Tempest?  Do you have any lightning or rain accumulation yet?  I've seem to have run into the same issue.  See https://github.com/peted-davis/WeatherFlow_PiConsole/issues/24

dwburger, 2020-09-09:
I can't answer your first question since I don't know how or where to check the logs.  Could you provide some guidance on that?  There are no lightning or rain accumulations.  I see on my Tempest iOS app that my history stopped in June.  I think I may need to check on that in my account.

falkyre, 2020-09-09:
How are you running the console (autostart or using wfpiconsole start)?  If you ssh to your raspberry pi, the software should be installed under /home/pi/wfpiconsole. Change your directory to there.  In that directory, there's a wfpiconsole.log.  If you are using the autostart, there won't be any exceptions in that log.  The way I found the exception was to run the console manually.  So I ran wfpiconsole stop and then wfpiconsole start.  When you do it that way, the software will log to your ssh session and you can see an exception getting thrown.

dwburger, 2020-09-09:
Yes, I've been using the autostart feature.  When I stopped and re-started wfpiconsole manually, I received the following messages in the terminal:

`pi@WeatherFlow:~/wfpiconsole $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_20-09-09_2.txt
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil, img_gif (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2(['window_egl_rpi'] ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
sdl2 - RuntimeError: b'Could not initialize EGL'
  File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/__init__.py", line 71, in core_select_lib
    cls = cls()
  File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line 152, in __init__
    super(WindowSDL, self).__init__()
  File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/window/__init__.py", line 981, in __init__
    self.create_window()
  File "/home/pi/.local/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line 290, in create_window
    self.get_gl_backend_name())
  File "kivy/core/window/_window_sdl2.pyx", line 224, in kivy.core.window._window_sdl2._WindowSDL2Storage.setup_window
  File "kivy/core/window/_window_sdl2.pyx", line 74, in kivy.core.window._window_sdl2._WindowSDL2Storage.die

[INFO   ] [Text        ] Provider: sdl2
2020-09-09 07:23:37-0700 [-] Log opened.
[CRITICAL] [App         ] Unable to get a Window, abort.
`

My RPi knowledge is limited so I'm not able to understand what those critical errors mean with regard to Windows.  Maybe you can???

dwburger, 2020-09-09:
My Tempest is not providing data so I need to figure that out first before continuing here.

dwburger, 2020-09-09:
My Tempest was offline.  I hadn't used the app for several weeks before trying wfpiconsole and wasn't familiar enough with the app to know the difference between online data and data coming from my local Tempest.  Sorry for the confusion!  I'll make sure my Tempest is working properly before moving forward with wfpiconsole.

dwburger, 2020-09-09:
Well...that's pretty embarrassing, but my wfpiconsole is working fine now with all panels showing correct data that coincides with data showing on my app.  I just needed to get my Tempest connected properly.  Again, thanks to falkyre for trying to help and sorry for contributing to confusion on this forum!

peted-davis, 2020-09-14:
Sorry for taking so long to respond - looks like my notifications stopped working.... Glad you have managed to get up and running again. I am going to close this issue as everything seems to be solved

closed 2020-09-14 by peted-davis