peted-davis/WeatherFlow_PiConsole #24: Tempest thread getting exception error

falkyre, 2020-09-09:
Hi,

Just started up my new Tempest and got this great console setup and going but I'm noticing that after the first start up, it seems like the temperature and wind speed values stop updating.  I am seeing that there is an exception thrown when there's no lightning strikes:

``` 
2020-09-08 21:47:49-0500 [-] Exception in thread Tempest:
2020-09-08 21:47:49-0500 [-] Traceback (most recent call last):
2020-09-08 21:47:49-0500 [-]   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
2020-09-08 21:47:49-0500 [-]     self.run()
2020-09-08 21:47:49-0500 [-]   File "/usr/lib/python3.7/threading.py", line 865, in run
2020-09-08 21:47:49-0500 [-]     self._target(*self._args, **self._kwargs)
2020-09-08 21:47:49-0500 [-]   File "/home/pi/wfpiconsole/lib/websocket.py", line 136, in Tempest
2020-09-08 21:47:49-0500 [-]     StrikeCount      = derive.StrikeCount(Strikes,StrikeCount,Device,Config,flagAPI)
2020-09-08 21:47:49-0500 [-]   File "/home/pi/wfpiconsole/lib/derivedVariables.py", line 553, in StrikeCount
2020-09-08 21:47:49-0500 [-]     if not math.isnan(todayStrikes[0]):
2020-09-08 21:47:49-0500 [-] UnboundLocalError: local variable 'todayStrikes' referenced before assignment
2020-09-08 21:47:49-0500 [-] 
```

Not sure if this is stopping the websocket from getting the weatherflow data from the REST api or not but I am definitely not getting updates (I see updates on the android app and the Tempest web page).

falkyre, 2020-09-09:
Further to this, I have 0 lightning strikes and no rain recorded yet.  So I'm getting the todayStrikes and TodayRain as being unreferenced local variables.  I've added a couple of lines in the derivedVariables.py that set those variables and that seems to have  fixed the exceptions and updates are happening (and no exceptions being thrown in the log)

I added 
```TodayRain = [NaN,'mm',NaN,Now]``` 

to the RainAccumulation function just after the setting of the timezone and the current time.

I also added  
```todayStrikes = [NaN,'count',NaN,Now]``` 
to the StrikeCount function, again just after the timezone and the current time.

peted-davis, 2020-09-14:
Sorry for taking so long to respond - my new issue notifications seem to have stopped working. The error definitely seems to be caused by the fact that your new Tempest has no rain accumulation and lightning strikes. I now have a Tempest to test with so I will look into this further

falkyre, 2020-09-15:
I removed the changes I made and it looks like things are working ok. I waited for an hour and things are updating with no issues. I guess I was impatient.  However I do now have rainfall listed and my tempest has been running for over a week now so not sure if that helped out. Not sure if you want to highlight this to new owners of the tempest or not, some people might thing your app isn't working if they aren't seeing updates right away.

closed 2020-09-15 by falkyre

peted-davis, 2020-09-15:
Great - glad to hear it is working! You definitely weren't being impatient - it's perfectly reasonable to expect the console to work right from the beginning. I have had a look at the code and I can see how to fix this, so hopefully new users won't see the same issue. Thanks for bringing it to my attention though!