peted-davis/WeatherFlow_PiConsole #25: Wfpiconsole shows Tempest offline after manual update

Death2Beavis, 2020-09-17:
Another apparent OS update issue. Exited wfpiconsole and ran apt update/upgrade. After reboot and the start of the console, the Tempest station shows offline and no data is gathered. Current wfpiconsole version 3.7. Checked logs and they did show a timeout connecting. Other network functions to the Pi are fine. Exited the wfpiconsole and reran wfpiconsole update (which is what seemed to fix it the last time). No luck, same result.

peted-davis, 2020-09-17:
This is a weird issue. Have you updated the OS successfully between this issue and your last (#16), or is this happening every time? Can you share the output of `wfpiconsole.log` here?

Death2Beavis, 2020-09-17:
Yes I have successfully updated (several times) before this update. I have
followed the protocol of exiting the console before the update and that (to
this point) has been just fine.

[INFO   ] [Logger      ] Record log in
/home/pi/.kivy/logs/kivy_20-09-17_1.txt
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil,
img_gif (img_ffpyplayer ignored)
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at
"/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not
docked
[INFO   ] [Text        ] Provider: sdl2
2020-09-17 10:03:30-0500 [-] Log opened.
2020-09-17 10:03:31-0500 [-] Starting factory
<__main__.WeatherFlowClientFactory object at 0x6fccc6b0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [Support     ] Starting twisted reactor
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts >
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range ABS X position is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 479
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range ABS Y position is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range position Y is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [GL          ] NPOT texture support is available
2020-09-17 10:03:33-0500 [-] Websocket connection open
2020-09-17 10:03:33-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] Unhandled Error
Traceback (most recent call last):
 File "/home/pi/.local/lib/python3.7/site-packages/twisted/python/log.py",
line 103, in callWithLogger
   return callWithContext({"system": lp}, func, *args, **kw)
 File "/home/pi/.local/lib/python3.7/site-packages/twisted/python/log.py",
line 86, in callWithContext
   return context.call({ILogContext: newCtx}, func, *args, **kw)
 File
"/home/pi/.local/lib/python3.7/site-packages/twisted/python/context.py",
line 122, in callWithContext
   return self.currentContext().callWithContext(ctx, func, *args, **kw)
 File
"/home/pi/.local/lib/python3.7/site-packages/twisted/python/context.py",
line 85, in callWithContext
   return func(*args,**kw)
--- <exception caught here> ---
 File
[... 132 lines cut ...]
line 1218, in consumeData
   while self.processData() and self.state !=
WebSocketProtocol.STATE_CLOSED:
 File
"/home/pi/.local/lib/python3.7/site-packages/autobahn/websocket/protocol.py",
line 1570, in processData
   fr = self.onFrameData(payload)
 File
"/home/pi/.local/lib/python3.7/site-packages/autobahn/websocket/protocol.py",
line 1665, in onFrameData
   self._onMessageFrameData(payload)
 File
"/home/pi/.local/lib/python3.7/site-packages/autobahn/twisted/websocket.py",
line 309, in _onMessageFrameData
   self.onMessageFrameData(payload)
 File
"/home/pi/.local/lib/python3.7/site-packages/autobahn/websocket/protocol.py",
line 601, in onMessageFrameData
   self.frame_data.append(payload)
builtins.AttributeError: 'NoneType' object has no attribute 'append'

2020-09-17 10:08:33-0500 [-] Websocket connection timeout
.
[INFO   ] [Support     ] Stopping twisted threads
[INFO   ] [Support     ] Shutting down twisted reactor

peted-davis, 2020-09-17:
OK, I don't think the issue is actually associated with the OS update. Instead I can see from the logs that it is failing because the incoming Websocket message is missing the `type` field:

`File "main.py", line 392, in WebsocketDecodeMessage`
`   Type = Msg['type']`
`builtins.KeyError: 'type'`

I have only ever seen this happen when the console tries and connects to the servers for the first time, hence you only see it when you run and update. I have made a bit of a hash of the error handling code here, and there is an obvious bug. If you try and connect a few times it might suddenly start working, and I will work on the bug this end.

Death2Beavis, 2020-09-17:
RGR. Will give that a swirl.

peted-davis, 2020-09-17:
OK, I've just pushed a new version that handles this error without crashing the console. Instead if `type` is missing from the Websocket message, it will print the message to the screen and then wait 60 seconds before attempting to reconnect. Can you update again with `wfpiconsole update` and then let me know how you get on? If it is crashing because `type` is missing, can you share the logs again so I can see what is going on?

Death2Beavis, 2020-09-17:
Ok, update applied. Its still not making the connection. I checked the INI
and the keys are in place.

  ===================================================
  New version detected
  Starting wfpiconsole configuration wizard
  ===================================================

  Required fields are marked with an asterix (*)

  API keys
  ---------------------------------
  No changes required

  Station and device IDs
  ---------------------------------
  No changes required

  Observation units
  ---------------------------------
  No changes required

  Display settings
  ---------------------------------
  No changes required

  "Feels Like" temperature cut-offs
  ---------------------------------
  No changes required

  Primary panel layout
  ---------------------------------
  No changes required

  Secondary panel layout
  ---------------------------------
  No changes required

  System settings
  ---------------------------------
  Updating version number to: v3.8

[INFO   ] [Logger      ] Record log in
/home/pi/.kivy/logs/kivy_20-09-17_13.txt
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil,
img_gif (img_ffpyplayer ignored)
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at
"/home/pi/.local/lib/python3.7/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not
docked
[INFO   ] [Text        ] Provider: sdl2
2020-09-17 14:04:18-0500 [-] Log opened.
2020-09-17 14:04:20-0500 [-] Starting factory
<__main__.WeatherFlowClientFactory object at 0x6fe40d50>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [Support     ] Starting twisted reactor
[INFO   ] [HIDMotionEvent] using <raspberrypi-ts >
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 799
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range ABS X position is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 479
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range ABS Y position is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range position X is 0 - 799
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] <raspberrypi-ts > range position Y is 0 - 479
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [GL          ] NPOT texture support is available
2020-09-17 14:04:22-0500 [-] Websocket connection open
2020-09-17 14:04:22-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] Unknown message type: {"status_code": 5,
"status_message": "AUTHORIZATION_REQUIRED - Rate limit exceed or your API
Key or Oauth2 Access Token missing"}
2020-09-17 14:04:22-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] Waiting 60 seconds to restart Websocket
connection
2020-09-17 14:05:33-0500 [-] Websocket connection closed
2020-09-17 14:05:33-0500 [-] Websocket connection retrying
2020-09-17 14:05:33-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] <twisted.internet.tcp.Connector instance at
0x6fdf98b0 disconnected IPv4Address(type='TCP', host='ws.weatherflow.com',
port=443)> will retry in 2 seconds
2020-09-17 14:05:33-0500 [-] Stopping factory
<__main__.WeatherFlowClientFactory object at 0x6fe40d50>
2020-09-17 14:05:35-0500 [-] Starting factory
<__main__.WeatherFlowClientFactory object at 0x6fe40d50>
2020-09-17 14:05:36-0500 [-] Websocket connection open
2020-09-17 14:05:36-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] Unknown message type: {"status_code": 5,
"status_message": "AUTHORIZATION_REQUIRED - Rate limit exceed or your API
Key or Oauth2 Access Token missing"}
2020-09-17 14:05:36-0500 [WeatherFlowClientProtocol
(TLSMemoryBIOProtocol),client] Waiting 60 seconds to restart Websocket
connection

peted-davis, 2020-09-17:
OK, so the Websocket is returning an error: `"AUTHORIZATION_REQUIRED - Rate limit exceed or your API
Key or Oauth2 Access Token missing"`. When you updated to version 3.7, did you generate a Personal Access Token (see the instructions here: https://community.weatherflow.com/t/weatherflow-piconsole/1933/1406?). If you didn't, follow the instructions and copy it into the `WeathFlowToken` field in the `.ini` file

Death2Beavis, 2020-09-17:
Ugh. No I did not. I regenerated the personal key and applied it. Works of
course.

[

peted-davis, 2020-09-17:
No worries - glad it is up and running! Hopefully this shouldn't happen again as it was the old API key that was causing the console to not connect properly.

closed 2020-09-17 by peted-davis