peted-davis/WeatherFlow_PiConsole #26: Cannot Enter SKY device ID

pmendiuk, 2020-09-29:
Hello,
I am trying to configure the piconsole but the configuration script keeps returning "SKY not found".  I am sure I have the correct ID.  I do have both an AIR/SKY and a Tempest but I do not want to use the Tempest sensor with the console.  The script asked if I had a Tempest and I have tried both yes and no.  The script does not find my Tempest nor my SKY sensors.

peted-davis, 2020-09-29:
Hi! Can you let me know your station IDs so I can look into this further? It might be as simple as confusing the device serial number with the device ID. If you go onto the Tempest app/web and click on `Settings` -> `Stations` -> [Station Name] -> `Status`, you should see all the Device and Station IDs listed (ignore the serial numbers).

pmendiuk, 2020-09-29:
Hi,

My station ID is 6655.

 

The IDs are 21824 and 21831.

 

Thanks,

Patrick

peted-davis, 2020-09-29:
Thanks - just so we are on the same page I have your station ID as 6655, your Air ID as 21824 and your Sky ID as 21823.

I have found the bug and it is an odd one. According to your station meta data, the elevation of the Sky above ground level is 0 m. Is this correct? Unfortunately 0 is equivalent to `False` and therefore the code thinks it has not been able to find the height of the Sky above ground level. The usual reason for this is because the device ID is wrong - hence the prompt. If set the value manually to 0.1 m, then the code works as expected.

If 0 m really is correct, you can fix this issue by temporarily setting the height to something greater than 0, restart the console and complete the configuration, and then reset the height back to 0 in the app/web. Then open the `wfpiconsole.ini` file and edit the `SkyHeight` key under `[Station]` to also read zero. I was also put a fix in place to account for zero height!

pmendiuk, 2020-09-29:
Yes that data is correct.  I just corrected the height for the SKY sensor but now I am getting “Access not authorized. Please re-enter your Weatherflow Personal Access Token*: “

 

Thanks,

Patrick

peted-davis, 2020-09-29:
> “Access not authorized. Please re-enter your Weatherflow Personal Access Token*: “

You need to double check the value of the Personal Access Token you are using. Once that is correct it should work as expected. You can also try deleting and regenerating a new one if the existing one just doesn't want to work. Do give it a couple of minutes between creating and token and using the token as sometimes I have seen a delay in new tokens being recognised.

pmendiuk, 2020-09-29:
Hi,

Thanks a lot!!!  It’s up and running.  I see the indoor fields are obviously not populated.  I am using MQTT for my WeeWX software to provide indoor temp and humidity.  Is that possible with this application?

 

Thanks,

Patrick

peted-davis, 2020-09-29:
> I see the indoor fields are obviously not populated.  I am using MQTT for my WeeWX software to provide indoor temp and humidity.  Is that possible with this application?

Unfortunately not. Out of the box only an indoor Air sensor is supported. The indoor temperature can be hidden using the console settings under the menu.

pmendiuk, 2020-09-29:
First of all I want to say how fantastic it is to have a console.  Thanks very much!!!

 

I think I will try to install the Paho MQTT client and try to “provide” that observation.

 

Best Regards,

Patrick

peted-davis, 2020-09-29:
Thanks for the kind words, and it'll be interesting to hear how you get on with MQTT. I have given some very brief instructions on how to incorporate other sensors in the console [here](https://community.weatherflow.com/t/weatherflow-piconsole/1933/1058?u=peter) which you might find useful.

closed 2020-09-29 by peted-davis