peted-davis/WeatherFlow_PiConsole #28: Today rain one number after the comma.

pjmpessers, 2020-12-31:
Is it possible to see the today rain with one number after the comma? Just like the yesterday rain does?

peted-davis, 2020-12-31:
Usually the number of decimal places depends on the amount of rain that has fallen. As the amount goes up, I restrict the number of digits to ensure it displays nicely without being too squished. What units are you using?

pjmpessers, 2020-12-31:
I use a raspberry pi 4 4gb with the PI touchdisplay 7 inch. The wfpiconsole is the 4.0.1

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

peted-davis, 2020-12-31:
Sorry, I meant what units do you use to display the rain accumulation? I assume either centimeters or millimeters.

pjmpessers, 2020-12-31:
Sorry. My mistake. It is in mm. The today rain and yesterday rain. Yesterday rain is in one number after the comma.

By the way. Have a great 2021.

[cid:image002.jpg@01D6DFAA.47614640]

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

peted-davis, 2020-12-31:
Not a problem. For millimeters when the total rainfall is greater than 10 mm, no number will be displayed after the comma. I assume this is what you are seeing. The Today rain is greater than 10, and the yesterday rain is less than 10. 

If you are comfortable editing the code it is easy to change this on lines 240-243 in the file `lib/observationFormat.py`. If you make a change though, be careful updating to new versions as this change will be overwritten

pjmpessers, 2020-12-31:
is it in line 240 making the 10 into 1?

peted-davis, 2020-12-31:
Not quite, but close. The easiest way to fix this is to replace the `{:.0f}` on line 243 with `{:.1f}`. Basically line 243 and 241 should be exactly the same.

pjmpessers, 2020-12-31:
Thank you. You helped a lot. When i'm back home i let you know if that works.

pjmpessers, 2021-01-01:
Thank you very much. It seems that this did the trick. I wait till the next rain

closed 2021-01-01 by peted-davis