peted-davis/WeatherFlow_PiConsole #29: "No module named distro"

wx-jc, 2020-12-31:
Just installed this on Ubuntu 18.04 running in a virtual environment, and got the following error when trying to start it:
Traceback (most recent call last):
  File "main.py", line 28, in <module>
    from lib     import config as configFile
  File "/home/jcook/wfpiconsole/lib/config.py", line 27, in <module>
    import distro
ModuleNotFoundError: No module named 'distro'

It showed that it got installed (check mark in the install status).

peted-davis, 2020-12-31:
Huh, that's weird. Try installing it manually with `python3 -m pip install distro` and then see if the console runs.

wx-jc, 2020-12-31:
Ok that seemed to work.  I tried installing it manually but it didn't work when I tried it, but with that command it did.  Thanks!

peted-davis, 2020-12-31:
Cool! Glad it's working. I'll have a dig around and see if I can work out why this didn't work out of the box.

closed 2020-12-31 by peted-davis