peted-davis/WeatherFlow_PiConsole #30: Touch screen blanks

erikb73, 2021-01-17:
Have successfully installed tempest weather station program to Pi, everything including touch on the touchscreen works beautifully, only prob is after a few minutes it goes dormant and screen blacks. Power cycling screen gets everything back up, but no touch, and it goes back to dormant after a few. Kind of had an idea that the display and interface would remain usable when I researched it. Am not a coder or programmer and I was hoping for some thoughts on this.

peted-davis, 2021-01-17:
This doesn't sound like normal behaviour. The console should remain usable for days on end, even if the screen goes into a power saving mode. Can you tell me what screen (make and model) you are using? When the screen goes blank you say that "power cycling screen" gets everything back up again. Does this mean that you don't reboot the Pi? You just turn the screen off and on again?

erikb73, 2021-01-24:
I bought the recommended Raspberry Pi 7” screen for the set up. I can get the display to come back up, but I have to power it off and on again, and it looses the “touch” interface when I do this. The data is current after the power cycle. May be I should update? Thanx for response. 🤙

erikb73, 2021-01-24:
Lebula LE070-01

peted-davis, 2021-01-25:
This sounds like an issue with the screen to be honest. If power cycling the screen brings the display back up, but you loose the touch, it sounds like it might be faulty. Is it too late to get a warranty replacement?

erikb73, 2021-01-27:
Am not sure this matters but the pi is going through the hdmi and power to the touchscreen. Am going to install the pi to the back of the touchscreen and see if that helps, if not I have a mouse installed, and I can just jog it to get get screen back on.

fpuigb, 2021-02-14:
It sounds to me that the display is going to sleep. 

You can try installing xscreensaver and disable the sleep. 

sudo apt-get install xscreensaver

Go to preference ---> screensaver.

You will see a screen saver main menu. In mode : section, select "disable screensaver" and close.

Reboot Rpi. It should work now.

closed 2021-08-16 by peted-davis