peted-davis/WeatherFlow_PiConsole #31: Wrong montly rain accumulation read out.

pjmpessers, 2021-02-03:
Something strange happend today. Yesterday and today there was a lot of rain. In the month total rainfall count you saw that yesterday and today was add up. Untill around 10:00 (GMT +1), where the yesterday totally rainfall count was removed and all what was left was the today totally rainfall count. (See photo)
Can this be fixed?

Thanks.

[image]

peted-davis, 2021-02-04:
I have been working on this bug over the last few days (fortunately it is quite rare), and should have an update in the next few days. You can try stopping and restarting the console to see if that fixes the display

pjmpessers, 2021-02-04:
Great to hear your working on it. I installed the beta, but didn't help. Also a restart didn't help. Today the month readout is zero. It isn't a big issue. I will wait patiently for the new update.👍🏻

peted-davis, 2021-02-14:
This should now be fixed in v4.0.3

pjmpessers, 2021-02-16:
Hello Peted-Davis.

For me it seems that the problem still exist. Yesterday it rained 1.7 mm and the month was showing 7.7 mm. But at night after a restart of my raspberry pi or at a given time (I don't know which?) It was back to 6.0 mm.

[image]

peted-davis, 2021-02-16:
I thought I had this sorted. I will take another look!

pjmpessers, 2021-02-16:
Thank you Peted-Davis.

peted-davis, 2021-02-16:
Can you share your station ID so I can look at you data in more detail?

pjmpessers, 2021-02-16:
There are now stupid questions. ;-) I'm already glad you help me. I changed some thinks for beter view from my couch. I update wfpiconsole with 'wfpiconsole update' command. After this installation i change some thinks in the files on a windows PC with notepad++ or on raspberry pi with geany. 

I will attached a file that shows in red text what I've exchanged, in which file and where.
My station id is https://tempestwx.com/station/32656/grid .

Let me know if you want more information.

[aanpassinng in wfpiconsole.docx](https://github.com/peted-davis/WeatherFlow_PiConsole/files/5992174/aanpassinng.in.wfpiconsole.docx)

peted-davis, 2021-02-17:
That all sounds good. I have just pushed out another update that will hopefully solve these issues. Can you run `wfpiconsole update` as usual and let me know if the bug is fixed? I will keep this issue open until we are certain it is working :-)

pjmpessers, 2021-02-17:
I installed the update and it looks very promising. After installation i see a change in the monthly readout from 6.0mm to 23.2mm. So that looks good. Now i need to wait for the next rainfall. ;-)

P.S. i wanted to say:"There are no stupid questions.", not "There are now stupid questions." Sorry for my bad English.

[image]

peted-davis, 2021-03-05:
Have you had any more rain to test the fixes I put in place?

pjmpessers, 2021-03-05:
It seems to work fine now. In the new month it was reset it to zero and yesterday there was a small rain that was calculated in the new month. So it looks fine.

peted-davis, 2021-03-06:
Great! I think this is solved now so I am going to close the issue, but do let me know if you notice any more bugs

closed 2021-03-06 by peted-davis

pjmpessers, 2021-03-06:
Thanks again.