peted-davis/WeatherFlow_PiConsole #32: Installer Failed on Pi 4 Model B

NickT-in-SD, 2021-02-12:
I ran the installation script per the instructions on this site, but the installer failed with the following:

"Error: unable to install Python module: autobahn[twisted]"
"error: can not find Rust Compiler"
"ERROR: Failed building wheel for cryptography"
"ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed correctly"

peted-davis, 2021-02-12:
Sorry you ran into issues. I have looked into this and unfortunately there have been some big changes to the Python cryptography module that I need to deal with. In the meantime, run this command in a terminal:

`python3 -m pip install cryptography==3.3.2`

and then try installing the PiConsole again. Hopefully this time everything will go smoothly.

NickT-in-SD, 2021-02-12:
Thank you! That worked.

peted-davis, 2021-02-14:
This should now be fixed in v4.0.3

closed 2021-02-14 by peted-davis