peted-davis/WeatherFlow_PiConsole #33: 4.0.3 update error

wx-jc, 2021-02-15:
[i] Updating WeatherFlow PiConsole Python modules...
  [✓] Updating Python package manager
  [✗] Updating Python module
  Error: Unable to update Python module: numpy

 /usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
ERROR: Could not find a version that satisfies the requirement numpy==1.20.1
ERROR: No matching distribution found for numpy==1.20.1

peted-davis, 2021-02-15:
Sorry you're having an issue. Can you let me know the hardware you are using, and your Python version?

wx-jc, 2021-02-15:
This is on an older intel computer running Ubuntu 18.04.  Python 3 version
is 3.6.9.

peted-davis, 2021-02-15:
Ok, I see the error I made. I'll work on a fix

peted-davis, 2021-02-15:
OK, this should now be fixed. Can you try running `wfpiconsole update` again please?

wx-jc, 2021-02-15:
That did the trick.  Thanks for the quick fix!

closed 2021-02-15 by wx-jc