peted-davis/WeatherFlow_PiConsole #34: no indoor temperature

djras1, 2021-03-01:
I have this all working great now.
But, there is no indoor temperature.
I have the Weatherflow Tempest and it has no indoor sensor.
I do have many different temperature sensors and could add a  DHT11 to the pi.

peted-davis, 2021-03-01:
This is expected behaviour. For users with first-generation Air and Sky modules from WeatherFlow, they have the option of bringing the Air module indoors to give an indoor temperature. This is no longer possible with the Tempest. You can either hide the indoor temperature under `settings` -> `display`, or feel free to edit the code to incorporate on of your own sensors.

closed 2021-03-01 by peted-davis

djras1, 2021-03-01:
Thanks!

⁣David Rasmussen ​

djras1, 2021-03-03:
I am close to implementing a DHT11/DHT22 local indoor temperature/humidity feature.

I have another question. Is there a way to get a screensave working with touch to wake option. I have tried using xscreensaver, but it won't blank the wfpiconsole display. 
I mainly only want a display from 6am to 10p. I could just put the pi power on a smart switch. But then would lose history of readings overnight. 

⁣David Rasmussen ​

peted-davis, 2021-03-05:
Hi David, it sounds like you are comfortable coding in Python so this module may be of interest: https://pypi.org/project/rpi-backlight/. It will allow you to control the backlight on the Raspberry Pi screen directly from the console/Python. You could create a function that is called at 6 am and 10 pm that turns on/off the backlight.

djras1, 2021-03-06:
Thanks!
That should work.
I found cron scheduling that can issue command scripts at specific times.

⁣David Rasmussen ​

djras1, 2021-03-06:
Hi Pete,
From a ssh shell I ran:
crontab -e
Add these 2 lines at end of the file to go bright at 6:30 am and dark at 10pm every day 

30 6 * * * /bin/sh /home/pi/bright.sh
0 22 * * * /bin/sh /home/pi/dark.sh

The .sh files are attached 

⁣David Rasmussen ​

peted-davis, 2021-03-06:
This is a good solution - thanks!

djras1, 2021-03-07:
What do ypu use to compile the code?
I am still working on the DHT22 sensor mod for indoor temperature (and humidity).

⁣David Rasmussen ​

peted-davis, 2021-03-08:
The code is written in Python so doesn't require compiling.

djras1, 2021-03-08:
Ah ok... I didn't put that together. 
Thanks 

⁣David Rasmussen ​

peted-davis, 2021-03-08:
No worries!