peted-davis/WeatherFlow_PiConsole #35: Last strike in minutes 1 digit instead of 2.

pjmpessers, 2021-03-13:
Hello.

I love the wfpiconsole. I use it every day. It was for the first time i got a strike. It is shown on the wfpiconsole. But when last strike is shown the minute under then is shown with one digit instead of two. For example i get: "last strike 16:2 instead of 16:02. Can this be corrected?

Thanks again.

peted-davis, 2021-03-13:
Sure - I will have a look at this. It sounds like a small glitch. Are you using the Official 7 inch Raspberry Pi touchscreen, or something different?

If possible, can you share a photo or screenshot showing the issue?

pjmpessers, 2021-03-13:
In the bedroom i have the original rpi 7 inch display. In the livingroom i hooked up a hdmi display. Both have the same glitch.

pjmpessers, 2021-03-13:
[image]
[image]

pjmpessers, 2021-03-13:
[image]

peted-davis, 2021-03-13:
Thanks for sharing the pictures. This is actually expected behaviour. If the hours or mins is less than 10, then they are not displayed with a leading zero. This is what it looks like on my display at the moment:

[image]

I understand why your display looks odd though, because on your display the hours and minutes are separated by a colon, which makes it look like a time. Are you running the latest version of the code (v4.0.4) as I actually removed the colon a number of versions back. The changes are made in the `wfpiconsole.kv` file. You may be overwriting this with the custom changes you are making to the display text sizes.

pjmpessers, 2021-03-13:
You are right. I made a adjustment. Didn't know that it would also change in days. I now understand your intention with this. In that case, thanks. There is no need for adjustment and we can close this comment.

Thanks again.

closed 2021-03-15 by pjmpessers