peted-davis/WeatherFlow_PiConsole #41: How to make shortcut on desktop for wfpiconsole?

pjmpessers, 2021-12-04:
Hello.

I can autostart the wfpiconsole, but sometimes it doesn't work after reboot with debian bullseye. But that is not a problem if i could start it with a shortcut on the desktop.
Could you explain in the readme how to make a working shortcut for wfpiconsole on the desktop? 

thanks.

10der, 2021-12-27:
mcedit /home/desktop/wfpiconsole.desktop

```
[Desktop Entry]
Comment=wf pi console
Terminal=true
Name=Htop
Exec=htop
Type=Application
Icon=/usr/local/bin/wfpiconsole
```

PS: mcedit is your fav Linux editor. you can use nano or vi, instead

peted-davis, 2021-12-30:
Following @10der approach, this is what you need to make a desktop shortcut:

Open a terminal and run `nano ~/Desktop/wfpiconsole.desktop`

Copy this into the empty file:

'
[Desktop Entry]
Comment=wfpiconsole
Terminal=true
Name=wfpiconsole
Exec=lxterminal -e wfpiconsole start
Type=Application
Icon=/usr/local/bin/wfpiconsole
'

Save the file with `ctrl-x` followed by `y` and `enter`. Now double click the new file that shows on the desktop and select `Execute`

pjmpessers, 2021-12-30:
Hello 10der and peted-davis.

Thank you both so very much. I couldn't find it out, so i will try this.
Again thanks guys. And have a great 2022.

peted-davis, 2021-12-30:
No worries @pjmpessers, I will add these instructions to the `readme`

closed 2021-12-30 by peted-davis