peted-davis/WeatherFlow_PiConsole #42: Default Window Position and Smallest Size

cefoster0, 2021-12-29:
This is amazing work! thank you so much! I'm trying to squeeze this into the lower left hand corner an existing PI display. I only have 640 x 360 available in that corner, to make room for other items like our family calendar. When I try and set the window size to that in the .ini, it doesn't seem to want to srink to that size. Other sizes seem to work so I believe I'm editing the correct .ini file. 

Also, is there an ability to set the default window position? I'd like to start the upper left hand corner of the window at 0,360 so it fits nicely in the corner. 

Thanks again for the hard work!

peted-davis, 2021-12-30:
Thanks for your kind words! The reason 640 x 360 isn't working is because the console has a hard-coded minimum size of 800 x 480. If you are happy editing the code directly, however, you can remove this limit.

You need to open `main.py` and comment out line numbers 74 and 75 (these set the minimum width and height).

You then need to find line 318 and replace the whole line with this: `self.scaleFactor = min(x/800, y/480)` (this scales the text and stops the different fields overlapping).

Let me know how you get on. If it works well, I could possibly remove this restriction in the next release

cefoster0, 2022-01-02:
thanks for pointing out where to look! commenting out those lines and editing did the trick and I think it resizes everything quite well! I mis-read your feedback earlier and so I have two screen shots, one with line 318 the same, one with the change. Change gets everything to have the proper fit and spacing but a little harder to read at a distance and burry on my TV set. But overall very nice. 

I quick google helped me find some kivy code to set the default position. I haven't gone through the effort of adding these to your config file format, let me know if it would be helpful to take a crack at that. 

 kivyconfig.set('graphics','position','custom')
    kivyconfig.set('graphics','left',-3)
    kivyconfig.set('graphics','top',330)

cefoster0, 2022-01-02:
[image]
[image]

peted-davis, 2022-03-17:
This is now the default behaviour in v22.3.2

closed 2022-03-17 by peted-davis