peted-davis/WeatherFlow_PiConsole #43: Station in low battery causes errors

cefoster0, 2022-01-02:
My Tempest station has been in mode 3 for several days as was new in box and only got one morning of sun before a week of rain and clouds here in the north east. After the station went into mode 3 a couple days ago, I noticed dashes across most of the display values. 

Running interactively, I saw the following errors:
   File "/home/pi/wfpiconsole/lib/derivedVariables.py", line 838, in MeanWindSpeed
2022-01-01 17:04:18-0500 [-]     AvgWind = [Sum/Length,'mps',Sum/Length,Length,Now]
2022-01-01 17:04:18-0500 [-] ZeroDivisionError: division by zero

It looks like since since my system is so new and has been in mode 2/3 for so long, I do not have valid wind data coming out of the api. 

I added a check. on 838 on the length value out of windups
 Length = len(windSpd)
            if Length > 0:

I also had to add another check and to line 896 
not math.isnan(windGust[0]):

That got most of the data that is populating working , except the 'feels like' temperature. which is odd as feels like shows up in the regular app/website. 

my station # is 47521 if you can peak via the API to see what's going on.

cefoster0, 2022-01-02:
Adding some photos. I reset the sizing code changes I made in the other ticket with no change. 
[image]
[image]
[image]

peted-davis, 2022-01-04:
Unfortunately this is a known bug and there is going to be a fix in the next, long overdue, update. Sorry it is causing you issues!

peted-davis, 2022-03-10:
This should now be fixed in the latest version (v22.3.1)

closed 2022-03-10 by peted-davis