peted-davis/WeatherFlow_PiConsole #45: Unable to update: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1

philbert2k, 2022-03-08:
I have been fighting with trying to update to the latest version.  After resolving the local cache / no internet issue I get the following:

      setup.py:374: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
         cy_ver = LooseVersion(cy_version_str)
       setup.py:377: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
         if cy_ver < LooseVersion(MIN_CYTHON_STRING):
       setup.py:381: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
         elif cy_ver > LooseVersion(MAX_CYTHON_STRING):
       warning: kivy/graphics/common.pxi:9:4: 'const_char_ptr' redeclared
       warning: kivy/graphics/common.pxi:23:4: 'size_t' redeclared
       error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for Kivy
Failed to build Kivy
ERROR: Could not build wheels for Kivy, which is required to install pyproject.toml-based projects

Verified that /usr/bin/arm-linux-gnueabihf-gcc does in fact exist:

root@rpi3wx:/home/pi# /usr/bin/arm-linux-gnueabihf-gcc
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.

peted-davis, 2022-03-08:
`root@rpi3wx:/home/pi`

Are you running as the root user? This is not required and might be what is causing the issue. Make sure you call the `wfpiconsole update` command from a regular, non-elevated terminal.

If that doesn't help, can you let me know what hardware you are running on and what OS.

philbert2k, 2022-03-08:
Trying it as a regular user (not 'sudo su') yields the same error/results.

HW: 
`Raspberry Pi 3 B+`

OS: 
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

If there is anything else I can provide, let me know.  Thanks!

peted-davis, 2022-03-08:
Can you run `cat /proc/device-tree/model` and let me know the ouput?

philbert2k, 2022-03-08:
> cat /proc/device-tree/model

Raspberry Pi 3 Model B Plus Rev 1.3

peted-davis, 2022-03-08:
Can you try running `sudo apt-get install python3-dev` and `sudo apt-get install build-essential` and then try the update again?

philbert2k, 2022-03-08:
> sudo apt-get install build-essential

`python3-dev is already the newest version (3.7.3-1).`
`python3-dev set to manually installed.`
-and-
`build-essential is already the newest version (12.6).`

peted-davis, 2022-03-10:
I'm really not sure what is going on here, sorry! I use a Pi3 (Rev 1.2 not 1.3) all the time for development and have never seen this. If you have a spare SD card, can you try flashing a new copy of Raspberry Pi OS and try installing the console from scratch. Do you still see the same issue?

philbert2k, 2022-03-12:
I will give that a try ... I am trying to figure out a possible fix before doing that... Reloading is always a pain.  

I think I have been able to get a little more details ... not sure if this helps any.

`building 'kivy.lib.gstplayer._gstplayer' extension
  creating build/temp.linux-armv7l-3.9/kivy/lib/gstplayer
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ikivy/include -I/usr/include/gstreamer-1.0 -I/usr/include/arm-linux-gnueabihf -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/python3.9 -c kivy/lib/gstplayer/_gstplayer.c -o build/temp.linux-armv7l-3.9/kivy/lib/gstplayer/_gstplayer.o
  arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-armv7l-3.9/kivy/lib/gstplayer/_gstplayer.o -L/usr/lib -lgstreamer-1.0 -lgobject-2.0 -lglib-2.0 -o build/lib.linux-armv7l-3.9/kivy/lib/gstplayer/_gstplayer.cpython-39-arm-linux-gnueabihf.so -lm
  arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-armv7l-3.9/kivy/lib/vidcore_lite/egl.o -L/opt/vc/lib -L/usr/lib -lbcm_host -lbrcmEGL -lbrcmGLESv2 -o build/lib.linux-armv7l-3.9/kivy/lib/vidcore_lite/egl.cpython-39-arm-linux-gnueabihf.so -lm
  arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-armv7l-3.9/kivy/graphics/svg.o -L/usr/lib -o build/lib.linux-armv7l-3.9/kivy/graphics/svg.cpython-39-arm-linux-gnueabihf.so -lm
   error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
  Building wheel for kivy (PEP 517) ... error
  ERROR: Failed building wheel for kivy
Failed to build kivy
ERROR: Could not build wheels for kivy which use PEP 517 and cannot be installed directly
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/req_command.py", line 180, in wrapper
    return func(self, options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 388, in run
    raise InstallationError(
pip._internal.exceptions.InstallationError: Could not build wheels for kivy which use PEP 517 and cannot be installed directly
Removed build tracker: '/tmp/pip-req-tracker-fsnyj5rt'
`

philbert2k, 2022-03-12:
Did a fresh install of Bullseye and I am getting the same exact thing :-(

peted-davis, 2022-03-12:
Ok, I will try and give this a go myself on my Pi3 and see what happens. I assume you are using the 32 bit version of Raspberry Pi OS (bullseye) with Desktop? 

You could always try using Raspberry Pi OS (legacy) in the meantime: https://www.raspberrypi.com/software/operating-systems/

philbert2k, 2022-03-12:
I

> Ok, I will try and give this a go myself on my Pi3 and see what happens. I assume you are using the 32 bit version of Raspberry Pi OS (bullseye) with Desktop?
> 
> You could always try using Raspberry Pi OS (legacy) in the meantime: https://www.raspberrypi.com/software/operating-systems/

Actually running the 64bit os.

peted-davis, 2022-03-13:
Ok, I would definitely give the 32 bit version a try. The console is entirely untested/unsupported on the 64 bit version

closed 2022-04-08 by peted-davis