peted-davis/WeatherFlow_PiConsole #47: 22.3.1 crash when Lightning button tapped

rtheil-growlink, 2022-03-10:
Just updated to 22.3.1 and the console crashes when I tap the lightning button the console crashes on Pi 4.

peted-davis, 2022-03-10:
I need more information to investigate. Can you share the details of the error that are printed to the screen?

rtheil-growlink, 2022-03-11:
[WARNING] [strikeDeltaT] 2022-03-11 06:25:20 - strikeTime is None
[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 242, in create_handler
     return eval(value, idmap), bound_list
   File "/home/pi/wfpiconsole/wfpiconsole.kv", line 840, in <module>
     'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \
 TypeError: '<' not supported between instances of 'NoneType' and 'int'

 During handling of the above exception, another exception occurred:

 Traceback (most recent call last):
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 697, in _apply_rule
     rctx['ids'])
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 247, in create_handler
     cause=tb)
 kivy.lang.builder.BuilderException: Parser: File "/home/pi/wfpiconsole/wfpiconsole.kv", line 839:
 ...
     837:    TitleField:
     838:        text:
 >>  839:            'Last Strike' if app.CurrentConditions.Obs['StrikeDeltaT'][4] == '-' else \
     840:            'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \
     841:            'Last Strike'
 ...
 TypeError: '<' not supported between instances of 'NoneType' and 'int'
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 242, in create_handler
     return eval(value, idmap), bound_list
   File "/home/pi/wfpiconsole/wfpiconsole.kv", line 840, in <module>
     'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \

 During handling of the above exception, another exception occurred:

 Traceback (most recent call last):
   File "main.py", line 498, in <module>
     wfpiconsole().run()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/app.py", line 950, in run
     runTouchApp()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 582, in runTouchApp
     EventLoop.mainloop()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 347, in mainloop
     self.idle()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 391, in idle
     self.dispatch_input()
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 342, in dispatch_input
     post_dispatch_input(*pop(0))
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/base.py", line 308, in post_dispatch_input
     wid.dispatch('on_touch_up', me)
   File "kivy/_event.pyx", line 709, in kivy._event.EventDispatcher.dispatch
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/behaviors/button.py", line 179, in on_touch_up
     self.dispatch('on_release')
   File "kivy/_event.pyx", line 705, in kivy._event.EventDispatcher.dispatch
   File "kivy/_event.pyx", line 1248, in kivy._event.EventObservers.dispatch
   File "kivy/_event.pyx", line 1132, in kivy._event.EventObservers._dispatch
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 57, in custom_callback
     exec(__kvlang__.co_value, idmap)
   File "/home/pi/wfpiconsole/wfpiconsole.kv", line 975, in <module>
     on_release: app.CurrentConditions.switchPanel(self)
   File "main.py", line 482, in switchPanel
     self.ids[button[1]].add_widget(eval(button[2] + 'Panel')())
   File "/home/pi/wfpiconsole/panels/lightning.py", line 40, in __init__
     super().__init__(**kwargs)
   File "/home/pi/wfpiconsole/panels/template.py", line 29, in __init__
     super().__init__(**kwargs)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/relativelayout.py", line 274, in __init__
     super(RelativeLayout, self).__init__(**kw)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/floatlayout.py", line 65, in __init__
     super(FloatLayout, self).__init__(**kwargs)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/layout.py", line 76, in __init__
     super(Layout, self).__init__(**kwargs)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/widget.py", line 361, in __init__
     rule_children=rule_children)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/uix/widget.py", line 465, in apply_class_lang_rules
     rule_children=rule_children)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 543, in apply
     rule_children=rule_children)
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 712, in _apply_rule
     e), cause=tb)
 kivy.lang.builder.BuilderException: Parser: File "/home/pi/wfpiconsole/wfpiconsole.kv", line 839:
 ...
     837:    TitleField:
     838:        text:
 >>  839:            'Last Strike' if app.CurrentConditions.Obs['StrikeDeltaT'][4] == '-' else \
     840:            'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \
     841:            'Last Strike'
 ...
 BuilderException: Parser: File "/home/pi/wfpiconsole/wfpiconsole.kv", line 839:
 ...
     837:    TitleField:
     838:        text:
 >>  839:            'Last Strike' if app.CurrentConditions.Obs['StrikeDeltaT'][4] == '-' else \
     840:            'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \
     841:            'Last Strike'
 ...
 TypeError: '<' not supported between instances of 'NoneType' and 'int'
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 242, in create_handler
     return eval(value, idmap), bound_list
   File "/home/pi/wfpiconsole/wfpiconsole.kv", line 840, in <module>
     'Strike Detected!' if app.CurrentConditions.Obs['StrikeDeltaT'][4] < 360 else \

   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 697, in _apply_rule
     rctx['ids'])
   File "/home/pi/.local/lib/python3.7/site-packages/kivy/lang/builder.py", line 247, in create_handler
     cause=tb)

peted-davis, 2022-03-11:
Thanks for the error log, I'll take a look into it. Can you share your Station ID as well so I can look at your data on the web?

rtheil-growlink, 2022-03-11:
Station ID 65150

closed 2022-03-28 by peted-davis

rtheil-growlink, 2022-03-29:
Verified working in 22.3.3. Thanks!