peted-davis/WeatherFlow_PiConsole #49: Updating to v22.3.2 fails.

mbeilste, 2022-03-11:
Getting an error on a few of the objects during the update.

[image]

peted-davis, 2022-03-11:
This looks like you have some corrupted files on your system. I would suggest that you take a backup of your `wfpiconsole.ini` file and then delete the entire PiConsole directory with `rm -rf ~/wfpiconsole/`

Fetch a new copy by running the install script

mbeilste, 2022-03-16:
Thanks!  That worked and got me updated.  I am now getting this when I try to start it.
[image]

mbeilste, 2022-03-17:
That did it.  Thank you sir!  Much appreciated!

closed 2022-03-17 by mbeilste