peted-davis/WeatherFlow_PiConsole #50: Update / fresh install to 22.3.2 freezes at Kivy install

digitaltree515, 2022-03-13:
My WFpiconsole runs on a Rpi A+, running Debian Buster.  When trying to update, at the Kivy step the whole system locks up and never progresses further (I've let it sit for 10 hours because I know Kivy takes some time).  Thinking maybe it was a bad OS install, I installed Buster on a wholly new SD card and attempted a fresh install of WFpiconsole.  Same results: whole system freezes up at Kivy and never progresses further.

peted-davis, 2022-03-13:
The issue here is likely the amount of memory available on your Pi. Installing Kivy takes a lot of memory, and if that plus you swap space is completely full, the install will freeze. 

Someone has seen this issue before and wrote about their solution here: https://community.weatherflow.com/t/weatherflow-piconsole/1933/2397

digitaltree515, 2022-03-13:
Thank you. I didn't come across any solutions when searching and I've been working on it for a few days. I'll give that a try in the morning. Thanks for being quick on the response!

closed 2022-05-15 by peted-davis