peted-davis/WeatherFlow_PiConsole #51: error: XDG_RUNTIME_DIR not set in the environment.

MCancellier, 2022-03-18:
I am running Buster on a Pi4 and just updated all the packages and wfpiconsole with the following string:

sudo apt update && sudo apt full-upgrade -y && sudo apt autoremove -y && sudo wfpiconsole update

When I attempt to start up the console, I get the following error:

error: XDG_RUNTIME_DIR not set in the environment.
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
sdl2 - RuntimeError: b'Could not initialize EGL'
  File "/root/.local/lib/python3.7/site-packages/kivy/core/__init__.py", line 70, in core_select_lib
    cls = cls()
  File "/root/.local/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line 152, in __init__
    super(WindowSDL, self).__init__()
  File "/root/.local/lib/python3.7/site-packages/kivy/core/window/__init__.py", line 982, in __init__
    self.create_window()
  File "/root/.local/lib/python3.7/site-packages/kivy/core/window/window_sdl2.py", line 289, in create_window
    self.get_gl_backend_name())
  File "kivy/core/window/_window_sdl2.pyx", line 226, in kivy.core.window._window_sdl2._WindowSDL2Storage.setup_window
  File "kivy/core/window/_window_sdl2.pyx", line 74, in kivy.core.window._window_sdl2._WindowSDL2Storage.die

x11 - ModuleNotFoundError: No module named 'kivy.core.window.window_x11'
  File "/root/.local/lib/python3.7/site-packages/kivy/core/__init__.py", line 62, in core_select_lib
    fromlist=[modulename], level=0)

[INFO   ] [Text        ] Provider: sdl2
 Traceback (most recent call last):
   File "main.py", line 516, in <module>
     wfpiconsole().run()
   File "/root/.local/lib/python3.7/site-packages/kivy/app.py", line 949, in run
     self._run_prepare()
   File "/root/.local/lib/python3.7/site-packages/kivy/app.py", line 919, in _run_prepare
     root = self.build()
   File "main.py", line 190, in build
     self.setScaleFactor(self.window, self.window.width, self.window.height)
 AttributeError: 'NoneType' object has no attribute 'width'

peted-davis, 2022-03-18:
Can you let me know what screen hardware you are using?

Also, what is the value of the `Hardware` key under the `[System]` section of the `wfpiconsole.ini` file?

MCancellier, 2022-03-18:
I'm using the 7 inch Raspberry Pi touch screen.

Hardware = Pi4

peted-davis, 2022-03-19:
Was the console working before you tried the update?

MCancellier, 2022-03-19:
Yes. This is the first time I've had an issue after a year plus of running flawlessly. 

I normally run updaters once a week, but had been out of town. This cycle there was a significant number of Raspberry Pi OS updates as well as your update.

peted-davis, 2022-03-19:
OK, it's good to hear it has been working before but clearly something during the update of the Raspberry Pi OS and the latest version of the console has gone wrong. I have just tested a fresh installation of Legacy version of Raspberry Pi OS that's based on Buster on one of my Pi4s and everything is working as expected. I think the best solution here is to probably re-flash your SD card and hope that fixes everything.

MCancellier, 2022-03-19:
I'll give it a try!

closed 2022-04-08 by peted-davis