peted-davis/WeatherFlow_PiConsole #52: Crash after tap lightning

KrunchMuffin, 2022-03-26:
**Describe the bug**
New station. Clicked lighting tab and crashed.

**To Reproduce**
Steps to reproduce the behavior:
Tapping lightning tab with new station.

**Screenshots**
[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: [Raspberry Pi OS Bullseye]
 - Hardware: [Pi4]
 - Console Version: [22.3.2]

**Additional context**
I know Python, but not Kivy. I was going to fix it but figured I'd post. 
I don't know if you can do it, but I think it should be `app.CurrentConditions.Obs['StrikeDeltaT'][4] is not None and...`

peted-davis, 2022-03-26:
Thanks for the report. You're right that I need to check for the `None` condition. I will get this fixed in the next version

peted-davis, 2022-03-28:
This should now be fixed in v22.3.3

closed 2022-03-28 by peted-davis