peted-davis/WeatherFlow_PiConsole #55: difficult to see rain intensity gauge with TFT monitor.

pjmpessers, 2022-04-07:
Hello.

Still love the wfpiconsole, but with a TFT monitor it is difficult to see the rain intensity gauge. This only has a good vissibillity when i'm standing right in front of the monitor. 

I worked around this in the previous versions by making a new raingauge.png, but this doen't work anymore in the newer version.  I named my own reworked raingauge.png rainintensity_hR.png and rainintensity_IR.png and placed it in /home/pi/wfpiconsole/icons/misc, but i then only saw a white block. 

Is there a way to make the raingauge less transparant?

Thanks.

peted-davis, 2022-04-08:
This should work. Make sure the icons are called `rainIntensity_lr.png` and `rainIntensity_hr.png` (the `I` in `Intensity` is capitalised`).

pjmpessers, 2022-04-08:
It does work. But the problem is the transparency. I prefer to have no transparency. When you look at the monitor at a angle it is very difficult to the raingauge. I would like to know if there is a possibility to change the transparency of the raingauge graphic?

pjmpessers, 2022-04-08:
That is great. Thank you.👍🏻

peted-davis, 2022-04-10:
I've uploaded the files to WeTransfer and you should be able to download the files here: https://we.tl/t-GkHKrWsXNs

pjmpessers, 2022-04-10:
Thanks for your support, Peted-davis. appreciate it. Is this a already modified file, or is it still necessary to modify?

peted-davis, 2022-04-12:
They still need editing. I'll try and send edited versions in a couple of days if that would be useful?

pjmpessers, 2022-04-12:
I think i got it. Now it is only a matter of waiting if it worked.

closed 2022-05-15 by peted-davis