peted-davis/WeatherFlow_PiConsole #56: hangs after shutdown on rebbot

VE3KH, 2022-04-11:
Hi ... I have a Ras Pi 3B with 7" touchscreen ... I am trying to get the wfpiconsole running ... I own a Tempest.  I have followed all the install instructions carefully and each time I get it all installed and I can see the wfpiconsole working nicely (3 times now LOL), but after I shutdown and re-apply power the ras pi seems to hang ... one time it complained about not finding a conifg.ini (as I recall), the latest time (now) it says "end Kernal panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0). Any help to resolve would be greatly appreciated :) ... Thanks, Kevin

peted-davis, 2022-04-12:
This sounds like your Raspberry Pi has a significant problem when trying to boot the operating system. I'm afraid I don't have solution for you, but my instinct would be that there is something wrong with the SD card you are using. It sounds like it is getting corrupted when the Pi is shutdown, and then it cannot be mounted or files are missing when you try to reboot. Try using a new SD card from a reputable brand and see if that fixes your issue

VE3KH, 2022-04-12:
Hi Pete

 

Thanks for that educated guess … much appreciated … I was starting to think the same thing, but I only had a 128 and 256 SD laying around, so I ordered a few more SD cards from Amazon yesterday … hopefully a do over with a new SD will fix the problem.

 

Thanks, Kevin

VE3KH, 2022-04-15:
Hi Pete

It seems it was a bad SD Card ... things are working fine now.

I have 2 questions.
1) On startup I need to click maximize to get the wfpiconsole to fit properly on the 7" screen ... is there a way to get it to scale properly without doing a maximize?
2) I see 2 mouse pointers (one black and one white) one lags the other while moving ... is that normal?

Thanks for your help and all the fine work you did developing this App and sharing :)

Thanks, Kevin

peted-davis, 2022-04-16:
Great! Good to hear the new SD card fixed your issues.

In answer to your questions, are you running on a Pi4? If so, have you followed these instructions from the `README`:

For those of you running a Raspberry Pi 4 an extra step is needed to get the console running perfectly. The `Task Bar` panel on the Raspberry Pi desktop needs to be hidden or else it will displace the console on the screen. There are two options to achieve this. First right click on the `Task Bar` panel on the Raspberry Pi desktop and select Panel settings. Select the Advanced tab. Then either:

Un-tick `Reserve space, and not covered by maximised windows`, or

Tick `Minimise panel when not in use` and set `Size when minimised` to 0 pixels

Also, make sure that under the `Display`section of the `Settings` that `Cursor` is set to off, and `fullscreen` is set to on. This can also be done by editing the `wfpiconsole.ini` file directly

VE3KH, 2022-04-16:
I am running a Pi 3B+ (I think) ... so I originally skipped those instructions.
I did set the Task Bar to hidden when not in use and 2 pixels wide.
I will try those other settings today
Thank You

peted-davis, 2022-04-16:
Ok, if you're definitely running a Pi3 you can skip those instructions. Try the other two suggestions and let me know how you get on. If it still isn't right, can you send a photo of how the console looks? There shouldn't be any problems getting it to startup full screen

closed 2022-05-15 by peted-davis