peted-davis/WeatherFlow_PiConsole #57: Indoor and Custom Panels

ntableman, 2022-05-25:
Hi

First this thing is so cool, well done! I love it.

I wanted to poke around with doing 2 things:

1 - filling in the existing data structures for inTemp, min and max - it seems your code is expecting a web socket message to do this. Do you have a sense of how I can modify your event model to poll a url for JSON? I am not overly familiar with this .kv framework...but if you have some idea of where to start I am happy to poke at it.

2 - custom panel - kind of the same thing - I am trying to grab JSON from a url and put in data - from all my custom sensors. I have tried something like calling that url and parsing the JSON in the cusomPanel.py, but it doesn't get called more than once and then I need to extend your base data structure which would make updating from your codebase harder. Again something to do with the .kv framework - do I need to setup an event model in the customPanel.py to poll repeatedly? Once I do that how do I pass the data to the .kv file when it is no in the superclass?

I write in all sorts of languages so I can figure this out (just newish to python, believe it or not) so this is more about what the conventions and framework does.

I'd love to share my shell of customPanel that polls URLs as I am sure other people have ideas in this area.

Thanks!

peted-davis, 2022-05-30:
1. You don't have to use a websocket to fill in the existing data structure - that's just the method I use to pull the data from the WeatherFlow servers. The .kv framework predominantly handles the graphically layout, and ensures changes in the data structure are automatically updated on the display. It will be relatively straight forward to poll a URL for a JSON. You just need to write a function that does the polling in a non-blocking manner which then updates the fields of the existing data structure (which you can pass as a input to your polling function). You will then want to use `Clock.schedule` to call your function at a set interval. You can seem some examples of this on lines 443-452 of `main.py`.

2. You can achieve the same in `customPanel.py`, but you will probably need to add import statements for `Clock.schedule`. You should be able to schedule the calling repeatedly by adding `Clock.schedule` in an additional `__init__` method in your custom panel class. You can also import `from Kivy import App` and use the `App.get_running_app()` method to access the existing data structure held in the app class. Something like `App.get_running_app().CurrentConditions.Obs`

disruptivepatternmaterial, 2022-06-14:
Thanks for the pointers! I have one new panel barely working and need some more time to focus on it. I was admittedly confused by the event model, but now I see it doesnt have to be all WS's. I'll happily share the panel when I get it working and it isnt embarrassing code mess :) if it might help others to show indoor and other data they may get from other sources.

closed 2023-03-05 by peted-davis