peted-davis/WeatherFlow_PiConsole #61: Error: Unable to install Kivy Python library

alexbromberek2020, 2022-06-15:
have a PI 3B with a new install of Debian Buster

[i] Kivy Python library installation check (will be installed)
  [✗] Installing Kivy Python library
  Error: Unable to install Kivy Python library

 Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/kivy/kivy/archive/2.0.0.zip
  Using cached https://github.com/kivy/kivy/archive/2.0.0.zip
ERROR: Exception:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/cli/base_comma                            nd.py", line 167, in exc_logging_wrapper
    status = run_func(*args)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/cli/req_comman                            d.py", line 205, in wrapper
    return func(self, options, args)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/commands/insta                            ll.py", line 342, in run
    reqs, check_supported_wheels=not options.target_dir
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/resolver.py", line 75, in resolve
    collected = self.factory.collect_root_requirements(root_reqs)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/factory.py", line 508, in collect_root_requirements
    requested_extras=(),
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/factory.py", line 473, in _make_requirement_from_install_req
    version=None,
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/factory.py", line 220, in _make_candidate_from_link
    version=version,
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/candidates.py", line 297, in __init__
    version=version,
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/candidates.py", line 161, in __init__
    self.dist = self._prepare()
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/candidates.py", line 230, in _prepare
    dist = self._prepare_distribution()
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/resolution/res                            olvelib/candidates.py", line 302, in _prepare_distribution
    return preparer.prepare_linked_requirement(self._ireq, parallel_builds=True)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/operations/pre                            pare.py", line 428, in prepare_linked_requirement
    return self._prepare_linked_requirement(req, parallel_builds)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/operations/pre                            pare.py", line 479, in _prepare_linked_requirement
    hashes,
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/operations/pre                            pare.py", line 165, in unpack_url
    unpack_file(file.path, location, file.content_type)
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/utils/unpackin                            g.py", line 240, in unpack_file
    unzip_file(filename, location, flatten=not filename.endswith(".whl"))
  File "/home/pi/.local/lib/python3.7/site-packages/pip/_internal/utils/unpackin                            g.py", line 115, in unzip_file
    zip = zipfile.ZipFile(zipfp, allowZip64=True)
  File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
    self._RealGetContents()
  File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

peted-davis, 2022-06-16:
Thanks for the report. I am away travelling at the moment, but will look into it as soon as I am back

alexbromberek2020, 2022-06-17:
I was able to install Kivy Python library  when i ran it as sudo -i

the next error i got was 

  [✗] Updating Kivy configuration for touch screen
  Error: Unable to update Kivy configuration for touch screen

 Traceback (most recent call last):
  File "python_command", line 9, in <module>
    with open('/home/pi/.kivy/config.ini','w') as configfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.kivy/config.i                                                                                                             ni'

i reran it and i was able to get it install

alexbromberek2020, 2022-06-17:
I was able to get it installed but I get this error? [INFO   ] [Websocket   ] 2022-06-17 17:09:11 - Opening connection
Unable to connect to X server

and thin I do a wfpiconsole autostart-enable and a reboot it works 

pi@piweather1:~ $ wfpiconsole start
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_22-06-17_6.txt
[INFO   ] [Kivy        ] v2.0.0
[INFO   ] [Kivy        ] Installed at "/home/pi/.local/lib/python3.7/site-packag                                                                              es/kivy/__init__.py"
[INFO   ] [Python      ] v3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 186 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffp                                                                              yplayer ignored)
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'OpenGL ES 2.0'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'VideoCore IV HW'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <b'OpenGL ES GLSL ES 1.00'>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Websocket   ] 2022-06-17 17:09:11 - Opening connection
Unable to connect to X server
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 4096
[INFO   ] [HIDMotionEvent] using <WaveShare WS170120>
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 4096
[INFO   ] [HIDMotionEvent] <WaveShare WS170120> range ABS X position is 0 - 4096
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 4096
[INFO   ] [HIDMotionEvent] <WaveShare WS170120> range ABS Y position is 0 - 4096
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] <WaveShare WS170120> range position X is 0 - 4096
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] <WaveShare WS170120> range position Y is 0 - 4096
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [Websocket   ] 2022-06-17 17:09:12 - Connection open

peted-davis, 2022-06-17:
Glad to hear you managed to get it installed. `Unable to connect to X server` sounds like you are trying to connect to your Pi via SSH. I highly recommend using VNC instead as it can handle the graphics much better than an SSH connection

alexbromberek2020, 2022-06-21:
I don't know where to ask a question but what do I need for the Indoor air temperature (requires an indoor Air module)I don't see one on there website

alexbromberek2020, 2022-06-26:
and when i do a VNC it dose not show the application it just shows the desktop but the main display shows it

closed 2022-08-08 by peted-davis