peted-davis/WeatherFlow_PiConsole #62: Display is inverted, touch control is backwards

MABeatty1978, 2022-08-06:
**Describe the bug**
I have this running on a raspberry pi 3, attached to the pi is a 5" touch display.  Everything works fine with the raspberry pi.  When WeatherFlow PiConsole is running the touch/mouse is inverted on the y axis.  If I touch the screen in the top right, the mouse moves to the bottom right.  If I drag the mouse to the top, the mouse drags to the bottom.

peted-davis, 2022-08-07:
Sorry you are having issues. This seems to be a bug that occurs with some particular touch screens. I will do some investigating and see if I can find a fix. If you are handy with linux, this link may help you find a solution quicker:

https://raspberrypi.stackexchange.com/questions/79858/kivy-on-rpi-how-to-correct-rotated-touch-screen

peted-davis, 2022-08-08:
OK, I have done some testing and I might have a fix available.

Open a terminal on your pi and run this command to open the Kivy configuration file:

`nano ~/.kivy/config.ini`

Scroll or page down right to the bottom and you should see two lines under `[input]` that look like this:
```
mtdev_%(name)s = probesysfs,provider=mtdev
hid_%(name)s = probesysfs,provider=hidinput
```
You need to add an extra parameter (`param=invert_y=1`) on the end of these lines so that they look like this:
```
mtdev_%(name)s = probesysfs,provider=mtdev,param=invert_y=1
hid_%(name)s = probesysfs,provider=hidinput,param=invert_y=1
```
Does this fix the issue?

MABeatty1978, 2022-08-08:
That did not work, but your link got me in the right direction.  I ended up leaving the mtdev_% line as is and editied the hid_% line to be
```
hid_%(name)s = probesysfs,provider=hidinput,param=invert_y=0
```
It's working as expected now.  Thanks for your help

closed 2022-08-08 by MABeatty1978