peted-davis/WeatherFlow_PiConsole #63: piconsole does not work after update

markalan2074, 2022-08-08:
Had it working, then decide to do a fresh install of RaspberryOS in 64 bit, worked one time, did update and get this error               
  INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
 Traceback (most recent call last):
   File "/home/AKCmas/wfpiconsole/main.py", line 35, in <module>
     configFile.update()
   File "/home/AKCmas/wfpiconsole/lib/config.py", line 120, in update
     currentVersion = currentConfig['System']['Version']
   File "/usr/lib/python3.9/configparser.py", line 960, in __getitem__
     raise KeyError(key)
 KeyError: 'System'

pls advise

peted-davis, 2022-08-09:
Delete the `wfpiconsole.ini` file and the n try starting the console again. You'll need to re-enter your API keys and station details etc, but hopefully it'll fix your issue

markalan2074, 2022-08-09:
Thank you for your quick response, deleted file now I get this: 
 
  Traceback (most recent call last):
   File "/home/AKCmas/wfpiconsole/main.py", line 33, in <module>
     configFile.create()
   File "/home/AKCmas/wfpiconsole/lib/config.py", line 97, in create
     writeConfigKey(Config, Section, Key, default[Section][Key])
   File "/home/AKCmas/wfpiconsole/lib/config.py", line 504, in writeConfigKey
     print('  Adding ' + keyDetails['Desc'] + ': ' + str(Value))
 UnicodeEncodeError: 'latin-1' codec can't encode character '\u2019' in position 27: ordinal not in range(256)

markalan2074, 2022-08-10:
> Can you let me know your station ID?
>
> Please note that the console is not officially supported on the 64 bit version of Raspberry Pi OS. This may be causing a compatibility issue and triggering the issue you are seeing. If you can, I would recommend trying the 32 bit version as I have not seen this error before

My station id is 77271 and yes I Have thought about downgrading back to 32 bit but wasn’t sure that would fix the problem.  Much impressed by piconsole and plan to use it on as static display for my Tempest, advise if that is the solution.

Thanks Mark

peted-davis, 2022-08-12:
Yes, I think downgrading is probably the best solution. I've never seen this error before, and the big difference with your setup is the 64 bit vs 32 bit. Hopefully a downgrade will fix everything, but if not I will keep digging into it

markalan2074, 2022-08-12:
Ok I’ll let you know if that fixes it.

markalan2074, 2022-08-15:
Ok I installed 32 bit, updated the OS have a fresh install.  The only thing I've done is configure my preferences and only added a Firefox browser and Libre Office, no interfaces were enabled (SSH, I2c, etc) except serial console which is enabled by default. Copy the script install command and it installed with no problems, then rebooted and ran wfpiconsole update, updated successful, rebooted.  Ran the start command, entered the data, and got same error as reported before  "UnicodeEncodeError: 'latin-1' codec can't encode character '\u2019' in position 27: ordinal not in range(256)". However I did add incorrect data at one point (mixed up station ID with Device ID) did not know how to stop script so closed the terminal window. Tried again with correct data and that's when I got the error, so I'm sure I blew it up.  Any clue how to fix?

peted-davis, 2022-08-15:
OK, this wasn't what I was expecting! I have tried to recreate the issue on my test setup using your station details and I can't get the same behaviour. Everything works ok. I assume you are using 77271 for your station ID and 205991 for your device ID? Can you let me know what hardware you are using (Pi3 or Pi4) and which version of Raspberry Pi OS (latest or legacy)? If you are not sure, can you share the output of this command: `cat /etc/os-release`

markalan2074, 2022-08-15:
P4 with 8 mb latest os, I’m wondering if it at the weather flow end, maybe when wrong data entered it got hard coded on their system and now doesn’t match? Second attempt everything entered correctly but get error message.

peted-davis, 2022-08-16:
OK. Entering the wrong information really shouldn't cause an issue. The console changes nothing on the WeatherFlow end, and as soon as you delete an incorrect `wfpiconsole.ini` file, the console will have no memory of the incorrect information. 

All I can think at the moment is that this issue has been triggered by modifying some preference or installing Firefox browser or Libre Office. I know it is a pain, but can you try installing the OS again and then install the Pi console immediately before configuring any preferences or installing any other software? When you run through the configuration steps after the OS first boots up, don't change any language settings or locale settings. Just keep the defaults. If that works we can then try and pin down exactly what causes the issue.

markalan2074, 2022-08-16:
Ok thank you I know now it's at my end and I'll figure it out. You can close the issue, again thank you for the support.

peted-davis, 2022-08-17:
Ha! Weird huh? Glad you're back up and running though

closed 2022-08-17 by peted-davis