peted-davis/WeatherFlow_PiConsole #64: Screen Size on Launch

Raiku17, 2022-08-18:
**Describe the bug**
On launch of the program, a bottom portion of the screen is cut off, and the rest is slightly stretched as if it is trying to fit a larger resolution.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the application
2. Once it loads, it is stretched

**Screenshots**
[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS 32-bit
 - Hardware: Pi 4b with 7" Raspberry Pi Touch Screen
 - Console Version: 22.5.1

**Additional context**
I followed the steps regarding screen size. In the Raspberry Pi OS I updated the taskbar settings to have it hide at 0 pixels. 
In the application settings I tried turning off and on the Launch in Full Screen option, it did not appear to have an impact.
I can resolve it by right clicking on the application in the taskbar and choosing maximize. It will then appear normal. This does not stay, because if restarted/shut down, the issue persists.
I also checked the wfpiconsole.ini file and confirmed the screen resolution matches the 7" touch screen.

peted-davis, 2022-08-18:
Thanks for the report and sorry you're having issues.

Can you confirm that you are using the official 7 inch touchscreen that is plugged into the Pi with a ribbon cable (rather than HDMI), and that the `hardware` key in the `.ini` file is set to `Pi4`. 

Can you also share the contents of the `wfpiconsole.log` file once you have started the console. That'll show which display settings the console is using.

Raiku17, 2022-08-18:
Thank you! The issue was the hardware was set to `Pi3` I updated to `Pi4` and issue has been resolved.

closed 2022-08-18 by Raiku17

peted-davis, 2022-08-18:
Great! Glad it was an easy fix 🙂