peted-davis/WeatherFlow_PiConsole #67: Failed at install - pyOpenSSL

zuper83, 2022-11-29:
**Describe the bug**
Failed at install
 [✗] Installing Python module
  Error: Unable to install Python module: pyOpenSSL

Error text...

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyOpenSSL==21.0.0
  Using cached https://www.piwheels.org/simple/pyopenssl/pyOpenSSL-21.0.0-py2.py3-none-any.whl (55 kB)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from pyOpenSSL==21.0.0) (1.16.0)
Collecting cryptography>=3.3
  Using cached cryptography-38.0.4.tar.gz (599 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting cffi>=1.12
  Using cached https://www.piwheels.org/simple/cffi/cffi-1.15.1-cp39-cp39-linux_armv7l.whl (364 kB)
Collecting pycparser
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.21-py2.py3-none-any.whl (119 kB)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml): started
  Building wheel for cryptography (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Building wheel for cryptography (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [193 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-39
      creating build/lib.linux-armv7l-cpython-39/cryptography
      copying src/cryptography/fernet.py -> build/lib.linux-armv7l-cpython-39/cryptography
      copying src/cryptography/exceptions.py -> build/lib.linux-armv7l-cpython-39/cryptography
      copying src/cryptography/__about__.py -> build/lib.linux-armv7l-cpython-39/cryptography
      copying src/cryptography/utils.py -> build/lib.linux-armv7l-cpython-39/cryptography
      copying src/cryptography/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography
      creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat
      copying src/cryptography/hazmat/_oid.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat
      copying src/cryptography/hazmat/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat
      creating build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/ocsp.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/name.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/oid.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/general_name.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/base.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/extensions.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      copying src/cryptography/x509/certificate_transparency.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
      creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/_asymmetric.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/poly1305.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/_serialization.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/_cipheralgorithm.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      copying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
      creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/backends
      copying src/cryptography/hazmat/backends/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/backends
      creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings
      copying src/cryptography/hazmat/bindings/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings
[... 145 lines cut ...]
          pip install --upgrade pip

      and then retry package installation.

      If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.

      This package requires Rust >=1.48.0.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

**To Reproduce**
Steps to reproduce the behavior:
1. Clean install of Rasbian OS (32bit) Lite
2. Updated and upgraded Rasbian OS thru SSH.
3. Run curl -sSL https://peted-davis.github.io/wfpiconsole | bash
4. Error during install at  Error: Unable to install Python module: pyOpenSSL

**Hardware/OS/Version:
 - OS: Raspberry Pi OS Bullseye (NAME="Raspbian GNU/Linux", VERSION_ID="11", VERSION="11 (bullseye)", VERSION_CODENAME=bullseye)
 - Hardware: Pi3 B 8 Gb
 - Console Version: N/A tried to install latest version available.

peted-davis, 2022-11-30:
Run this command first and then try reinstalling:

`python3 -m pip install cryptography==37.0.4`

peted-davis, 2022-12-01:
Is the `Hardware` key in your `wfpiconsole.ini` set to `Pi3`?

zuper83, 2022-12-01:
Yes, Pi3 confirmed.

Only difference else is that

#dtoverlay=vc4-kms-v3d

in /boot/config.txt. Tried option with activate it, but no difference unfortunately.

vh
Markus Lindholm

peted-davis, 2022-12-01:
Ah, I see the issue. You've installed the Bullseye version of Raspberry Pi OS. For Pi3 you need to use the legacy (Buster) version. Unfortunately you'll need to wipe your SD card clean, install the legacy OS, run the extra command that fixed the first issue, and then run the installer again

zuper83, 2022-12-01:
Hi,

You are so correct, it´s clearly stated in readme! Case closed. I'll start over with Buster and my sincerely apology for wasting your valuable time.

peted-davis, 2022-12-03:
No problem! Glad we managed to get it fixed

closed 2022-12-03 by peted-davis