peted-davis/WeatherFlow_PiConsole #68: Wfpiconsole fails in Sager routines

ctgreybeard, 2022-12-03:
**Describe the bug**
I noticed that my weather screens were down this evening. Restarting it fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to start the weather console

**Screenshots**
None

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi Buster
 - Hardware: PI4
 - Console Version: 22.5.1

**Additional context**
Log shows error in Sager

[wfpiconsole.log](https://github.com/peted-davis/WeatherFlow_PiConsole/files/10145490/wfpiconsole.log)

Grandmasg, 2022-12-03:
Found a temp fix.. /home/pi/wfpiconsole/lib/sager.py", line 266
`
            #self.sager_data['METAR'] = Data.json()['data'][0]
            self.sager_data['METAR'] = ''
`
[image]

peted-davis, 2022-12-03:
Thanks for the report, I will look into this ASAP (probably Monday at the latest)

creas002, 2022-12-03:
Sager causing same issue for me as well.

[image]

bgiuriceo, 2022-12-03:
Same with me 
[image]

pjmpessers, 2022-12-03:
That code did work for me. Thank you. I can use wfpiconsole again. Great.
I will wait for the fix, but happy that i can use the console again.

KrunchMuffin, 2022-12-03:
So there is no data coming back from the METAR API, so instead of making it empty and I added an AND to check for length so when the API comes back it will work again.

```
 if checkwx_api.verify_response(Data, 'data') and len(Data.json()['data']):
```

peted-davis, 2022-12-03:
This should now be fixed in v22.21.1.

To update please run `wfpiconsole stop` followed by `wfpiconsole update`. Please make sure any local changes to the code or custom panels are backed up.

closed 2022-12-03 by peted-davis

creas002, 2022-12-03:
> This should now be fixed in v22.12.1.
> 
> To update please run `wfpiconsole stop` followed by `wfpiconsole update`. Please make sure any local changes to the code or custom panels are backed up.

Thanks, new version resolved issue for me.

pjmpessers, 2022-12-03:
Is there more changes in these codes? If not, is it possible to only update the sager.py?

peted-davis, 2022-12-04:
> Is there more changes in these codes? If not, is it possible to only update the sager.py?

There are some other small bug fixes scattered throughout the update, but the changes to fix this particular bug are all in `sager.py`

pjmpessers, 2022-12-04:
Thank you. I looked at the code which where new and only updated those. Works perfectly again. 👍🏻