peted-davis/WeatherFlow_PiConsole #69: Console crashed.

dethompson1958, 2022-12-03:
**Describe the bug**
A clear and concise description of what the bug is.
Console had crashed.  Tried to restart using "wfpiconsole start" and the UI flashed and nothing happened.  Tried rebooting the PI and trying again to start console and nothing happens.  Tried to attach screen shot of attempting to start from the terminal.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: [e.g. Raspberry Pi OS Bullseye]
 - Hardware: [e.g. Pi3]
 - Console Version: [e.g. 22.3.1]

**Additional context**
Add any other context about the problem here.

Grandmasg, 2022-12-03:
Same issue --> https://github.com/peted-davis/WeatherFlow_PiConsole/issues/68

philbert2k, 2022-12-03:
Here as well. Not sure if this could be related to server maintenance somewhere (based on the error) but will keep checking and watching here.

Kuuke, 2022-12-03:
Glad I'm not the only one ;)

OS: Raspberry Pi Bullseye
Hardware: RPI4
Python: 3.9
WFPI Console version: 22.5.1

Grandmasg, 2022-12-03:
Temp fix --> https://github.com/peted-davis/WeatherFlow_PiConsole/issues/68#issuecomment-1336097746

peted-davis, 2022-12-03:
Thanks for the report, I will look into this ASAP (probably Monday at the latest)

peted-davis, 2022-12-03:
Duplicate of #68

closed 2022-12-03 by peted-davis

dethompson1958, 2022-12-03:
Thanks for the quick fix.  All is working great after update.  Thanks again.