peted-davis/WeatherFlow_PiConsole #71: Crashing at last_24h call

philbert2k, 2022-12-04:
Not sure if this has any relation to the Sager issue or not, but just in the last 24-48 hours I have noticed that the console is crashing and requires intervention to restore it to the previous state.   For me, I believe I have a few issues.  This being one.  

The other issue(s) I have is that the settings seem to be ignored and there is also, when full screen, a "wfpiconsole" title bar of the window.  IE: Full Screen, Ignore Mouse, etc are ignored but selections retained.  

[image]

**Hardware/OS/Version (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye
 - Hardware: Raspberry Pi 3 Model B Plus Rev 1.3
 - Console Version: 22.12.1

philbert2k, 2022-12-04:
Update: Seems to be unrelated to the last_24h call (or so it seems...) as it crashed again with no additional info in the log after "Connection open"

peted-davis, 2022-12-04:
Sorry you're having issues. 

I see you are running on a Pi3. This is why the settings for fullscreen etc. are ignored. The Pi3 does not have a graphics card capable of supporting these features. 

You've also indicated that you are running the Bullseye version of Raspberry Pi OS. Unfortunately, the Pi3 is only compatible with the Buster (legacy) version of Raspberry Pi OS (see README). This is causing the taskbar that you are seeing. 

I suggest that you start again with the fresh SD card and the legacy version of Raspberry Pi OS. This will hopefully clear up all the issues you see.

bgiuriceo, 2022-12-04:
Thanks for the [working] fix. Now I know how much I rely on the program.....Thanks again.

philbert2k, 2022-12-04:
Thanks for the info.  I can live with some of that ... easier than starting everything over haha.  I will keep an eye on the main issue and see if it persists.  Things ran soo smooth for months and then this just started.  Maybe it's just time to update the hardware.

closed 2023-03-05 by peted-davis