peted-davis/WeatherFlow_PiConsole #72: Does not start correctly after updating from 22.5.1 to 22.12.1

Striken5, 2022-12-04:
**This is the info in the console window that appears**

> Exception in thread obs_st:
> Traceback (most recent call last):
>   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.7/threading.py", line 865, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/pi/wfpiconsole/lib/observationParser.py", line 154, in parse_obs_st
>     self.calcDerivedVariables(device_id, config, 'obs_st')
>   File "/home/pi/wfpiconsole/lib/observationParser.py", line 425, in calcDerivedVariables
>     self.derive_obs['peakSun']   = derive.peakSunHours(self.device_obs['radiation'],  self.derive_obs['peakSun'], device, self.api_data, >config)
>   File "/home/pi/wfpiconsole/lib/derivedVariables.py", line 1421, in peakSunHours
>     sunrise           = Observer.next_rising(ephem.Sun()).datetime().timestamp()
>   File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 439, in next_rising
>     return self._find_rise_or_set(body, start, use_center, +1, True)
>   File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 497, in _find_rise_or_set
>     % (body.name, self.date))

**I have comment out lines 439, 440, 443 and 444 of main.py because this error**

> Traceback (most recent call last):
>  File "main.py", line 521, in <module>
>     wfpiconsole().run()
>   File "/home/pi/.local/lib/python3.7/site-packages/kivy/app.py", line 949, in run
>     self._run_prepare()
>   File "/home/pi/.local/lib/python3.7/site-packages/kivy/app.py", line 919, in _run_prepare
>    root = self.build()
>   File "main.py", line 199, in build
>     self.screenManager.add_widget(CurrentConditions())
>   File "main.py", line 439, in __init__
>     self.app.astro.sunrise_sunset()
>  File "/home/pi/wfpiconsole/lib/astronomical.py", line 116, in sunrise_sunset
>     Sunrise = self.observer.next_rising(self.sun)
>   File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 439, in next_rising
>     return self._find_rise_or_set(body, start, use_center, +1, True)
>   File "/home/pi/.local/lib/python3.7/site-packages/ephem/__init__.py", line 497, in _find_rise_or_set
>     % (body.name, self.date))
> ephem.NeverUpError: 'Sun' is below the horizon at 2022/12/4 10:48:19

**Is this the fault?**
**Or is there another method for us who live in the far north?**

**It is only half of the wind info in winr rose and sager/forecast thats work when I start wfpiconsole**

[image]

peted-davis, 2022-12-05:
I don't think this is related to the update. There was no change to the astronomical code. Instead, it's probably related to the time of year and by coincidence it coincided with the update. 

Unfortunately, the code is not designed to handle the situation where the sun doesn't rise/set. It is a known limitation and is referenced in issue #. There is a work around in that issue, and I am currently working on getting the additional functionality incorporated into the code.

Striken5, 2022-12-05:
I have comment out lines 439, 440, 443 and 444 of main.py
Like in post #60. 
Then the screen looks like the one in the picture.
I dont understand why time of year should have anyting with temprature reading, preasure reading, wind readings and all the other readings, execpt sun and moon graph. All the other readings should come from the weather station/web connection?

peted-davis, 2022-12-05:
Time of year is responsible for causing the sunrise/sunset bug. The lack of weather data (which does come from the weather station) is a consequence of the sunrise/sunset bug. 

Did the workaround suggested in #60 work for you? If not, can you share your station ID so I can investigate further

Striken5, 2022-12-05:
After the workaround in #60 it starts the program. But only with wind reading.
Station id 53780
Line 439, 440, 443 and 444 is # out in main.py
sun and moon

peted-davis, 2022-12-05:
OK, I missed that the solar information is used elsewhere as well. You also need to comment out line 425 in `lib/observationParser.py`. That should get you back up and running with any sun/moon data until I get this bug fixed

Striken5, 2022-12-05:
👍👍👍
Now its runing 😁
Then you know the stationID for testing the sun moon code 😊

peted-davis, 2022-12-15:
Closing as a duplicate of #60

closed 2022-12-15 by peted-davis