peted-davis/WeatherFlow_PiConsole #73: IndexError: list index out of range

geewizard1, 2022-12-05:
RPI 3, Debian Buster OS, functional WeatherFlow Tempest station.

After entering both API keys and Station and device IDs, I get this:

 Traceback (most recent call last):
   File "main.py", line 33, in <module>
     configFile.create()
   File "/home/pi/wfpiconsole/lib/config.py", line 97, in create
     writeConfigKey(Config, Section, Key, default[Section][Key])
   File "/home/pi/wfpiconsole/lib/config.py", line 418, in writeConfigKey
     for Device in STATION['stations'][0]['devices']:
 IndexError: list index out of range

Not sure how to proceed with configuration.....

peted-davis, 2022-12-05:
What is your station and device ID?

geewizard1, 2022-12-05:
Station and device IDs
  ---------------------------------
  Please enter your Station ID*: 19002
  Do you own a TEMPEST?* [y/n] y
  Please enter your TEMPEST device ID*: 65067

peted-davis, 2022-12-05:
Are you sure you are using the correct station ID and that you have added the Tempest device to your station? The API response for station 19002 is blank:

`{"stations":[],"status":{"status_code":0,"status_message":"SUCCESS"}}`

geewizard1, 2022-12-05:
Ah!  I was using old values.  It's going now.  Thanks for the hints!

 Station and device IDs
  ---------------------------------
  Please enter your Station ID*: 80926
  Do you own a TEMPEST?* [y/n] y
  Please enter your TEMPEST device ID*: 214243
  Do you own an Indoor AIR?* [y/n] n

peted-davis, 2022-12-05:
Great! Glad it's fixed

closed 2022-12-05 by peted-davis