peted-davis/WeatherFlow_PiConsole #75: Going to menu results in a crash.

pjmpessers, 2022-12-06:
Hello Peted-Davis.

Still love your app. But have a crash when going to menu. Removed and clean installed and i can go to menu again. Great.

But when i place back the backup of wfpiconsole.kv, displayfield.kv, observationformat.py and the panellayout.kv the problem is back again. Do you know where the problem is coming from?

Thanks again if you could help me.

Traceback (most recent call last):
   File "kivy/properties.pyx", line 861, in kivy.properties.ObservableDict.__getattr__
 KeyError: 'selector_panel'
 
 During handling of the above exception, another exception occurred:
 
 Traceback (most recent call last):
   File "/home/pi/wfpiconsole/main.py", line 521, in <module>
     wfpiconsole().run()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/app.py", line 950, in run
     runTouchApp()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/base.py", line 582, in runTouchApp
     EventLoop.mainloop()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/base.py", line 347, in mainloop
     self.idle()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/base.py", line 391, in idle
     self.dispatch_input()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/base.py", line 342, in dispatch_input
     post_dispatch_input(*pop(0))
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/base.py", line 308, in post_dispatch_input
     wid.dispatch('on_touch_up', me)
   File "kivy/_event.pyx", line 709, in kivy._event.EventDispatcher.dispatch
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/uix/behaviors/button.py", line 179, in on_touch_up
     self.dispatch('on_release')
   File "kivy/_event.pyx", line 705, in kivy._event.EventDispatcher.dispatch
   File "kivy/_event.pyx", line 1248, in kivy._event.EventObservers.dispatch
   File "kivy/_event.pyx", line 1132, in kivy._event.EventObservers._dispatch
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/lang/builder.py", line 57, in custom_callback
     exec(__kvlang__.co_value, idmap)
   File "/home/pi/wfpiconsole/wfpiconsole.kv", line 100, in <module>
     on_release: Factory.mainMenu().open()
   File "/home/pi/.local/lib/python3.9/site-packages/kivy/uix/modalview.py", line 221, in open
     self.dispatch('on_pre_open')
   File "kivy/_event.pyx", line 709, in kivy._event.EventDispatcher.dispatch
   File "/home/pi/wfpiconsole/panels/menu.py", line 83, in on_pre_open
     self.ids.selector_panel.add_widget(station_selector())
   File "kivy/properties.pyx", line 864, in kivy.properties.ObservableDict.__getattr__
 AttributeError: 'super' object has no attribute '__getattr__'

peted-davis, 2022-12-15:
Sorry, missed this one! I went back over the update and I'd forgotten that I have made a bunch of changes to the structure of the code - in particular the `.kv` files that control the layout of the console. That's why when you add your backups back, you are seeing errors. 

There is a `CustomPanel` framework that allows you to design your own panels that will be update safe. You might want to consider looking at this route if you are making significant modifications to the layout. Otherwise I cannot guarantee that updates will not break your custom changes

pjmpessers, 2022-12-15:
P.S./ Can you tell me in wich file all the adjustments went what was previously in wfpiconsole.kv?

peted-davis, 2022-12-22:
Sorry - missed this. Most of the contents of `wfpiconsole.kv` have been moved to individual files in the `kvlang` folder

pjmpessers, 2022-12-22:
@peted-davis Thank you, Peted. Thanks by you i found it. . So i will make my adjustments. :-)

peted-davis, 2022-12-28:
Glad I could help!

closed 2022-12-28 by peted-davis